tmk/tmk_keyboard #168: SysReq key not being sent?

frioux, 2015-01-12:
I have a custom layout at http://github.com/frioux/tmk_keyboard, it's a fork of @cub-uanic's ergodox port.  I wanted to add a SYSREQ key and for some reason it never seems to send the right keycode.  On a regular keyboard, Alt+SysReq sends this: 

      KeyPress event, serial 41, synthetic NO, window 0x2400001,
          root 0x9e, subw 0x0, time 3955817, (574,832), root:(575,852),
          state 0x0, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
          XLookupString gives 0 bytes: 
          XmbLookupString gives 0 bytes: 
          XFilterEvent returns: False

      KeyPress event, serial 41, synthetic NO, window 0x2400001,
          root 0x9e, subw 0x0, time 3955817, (574,832), root:(575,852),
          state 0x8, keycode 107 (keysym 0xff15, Sys_Req), same_screen YES,
          XLookupString gives 0 bytes: 
          XmbLookupString gives 0 bytes: 
          XFilterEvent returns: False

      KeyRelease event, serial 41, synthetic NO, window 0x2400001,
          root 0x9e, subw 0x0, time 3955817, (574,832), root:(575,852),
          state 0x8, keycode 107 (keysym 0xff15, Sys_Req), same_screen YES,
          XLookupString gives 0 bytes: 
          XFilterEvent returns: False

      KeyRelease event, serial 41, synthetic NO, window 0x2400001,
          root 0x9e, subw 0x0, time 3955817, (574,832), root:(575,852),
          state 0x8, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
          XLookupString gives 0 bytes: 
          XFilterEvent returns: False

on my ergodox, Alt+SysReq sends this:

      KeyPress event, serial 35, synthetic NO, window 0x2a00001,
          root 0x9e, subw 0x0, time 4016519, (279,757), root:(280,777),
          state 0x0, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
          XLookupString gives 0 bytes: 
          XmbLookupString gives 0 bytes: 
          XFilterEvent returns: False

      KeyPress event, serial 35, synthetic NO, window 0x2a00001,
          root 0x9e, subw 0x0, time 4017255, (279,757), root:(280,777),
          state 0x8, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
          XLookupString gives 0 bytes: 
          XmbLookupString gives 0 bytes: 
          XFilterEvent returns: False

      KeyRelease event, serial 35, synthetic NO, window 0x2a00001,
          root 0x9e, subw 0x0, time 4017551, (279,757), root:(280,777),
          state 0x8, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
          XLookupString gives 0 bytes: 
          XFilterEvent returns: False

      KeyRelease event, serial 35, synthetic NO, window 0x2a00001,
          root 0x9e, subw 0x0, time 4017559, (279,757), root:(280,777),
          state 0x8, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
          XLookupString gives 0 bytes: 
          XFilterEvent returns: False

I noticed that no one else ever uses the SYSREQ key, at least in the default tmk
repo.  Is it possible that maybe it's incorrectly defined?

tmk, 2015-01-12:
Also I for one doesn't use 'System Request' key for anything and honestly don't know much about how the key is recognized in Xorg, Linux and other OS's.

Which keycode are you using for 'System Request'? KC_SYSREQ or KC_PSCREEN? (Where is your keymap file?)
What physical keys are you pressing on your Ergodox?
And what do you use the key for? For some Xorg apps or 'Magic SysRq'?

First, it seems that keycode KC_SYSREQ(0x9A) is not used in Linux and Windows at least. I guess that in PC keyboards 'System Request' key is probably combination of 'Alt' and 'PrintScreen'.

Read resouces on the net, try them and report back! I'd like to know your result.

http://en.wikipedia.org/wiki/Magic_SysRq_key
http://en.wikipedia.org/wiki/System_request

tmk, 2015-01-12:
And how about using 'showkey' on console instead of 'xev'? I think you'll be able to get more 'raw' keycodes with that.

frioux, 2015-01-13:
I am using KC_SYSREQ, but I'll try KC_PSCREEN in a minute.

The keymap is at https://github.com/frioux/tmk_keyboard/blob/e89161b3b2e5ec20138d2bd0c0d32df1b51f8935/keyboard/ergodox/keymap_frew.h; I was trying to add it to the left hand thumb cluster, the bottom right key there.

I am trying to use it for Magic SysRq.

I'll also try with showkey too, though xev did show the SysReq on a different keyboard.

Making new build now!  Will post if KC_PSCREEN works.

frioux, 2015-01-13:
You were right, KC_PSCREEN made it work!  Thanks!

closed 2015-01-13 by frioux