tmk/tmk_keyboard #173: Removal of projects or separation of core library and project

tmk, 2015-01-19:
I have plan to remove some of projects which I can't mainatain from repository, or move projects to another(each own) repository.

1) Most of converter projects will continue to be updated by hasu/tmk(me) because I still own keyboards needed to test firmware. Meanwhile most of keyboard projects can't be tested. hasu/tmk owns keyboard pojejcts; gh60, hbkb, hhkb, infinity, onekey and onekey_mbed. But they may be separated from core library.

2) If contributor of the project has a repository it should be center of the project. README of tmk_keyboard just has a link to the repository and tmk_keyboard has no code for the project.

xauser, 2015-01-19:
I can offer to test the following keyboard projects: kitten_paw, kmac, lightpad, lightsaber, nerd and phantom.

tmk, 2015-01-19:
Thanks.
Probably I'll contact you when I make decision about this.

p3lim, 2015-01-27:
What about having each keyboard/converter it's own repository, then reference to the git repository within the directories?

I do something like this in many of my projects (although they are completely irrelevant to this firmware), here's an example: https://github.com/p3lim-wow/Molinari/tree/master/embeds

p3lim, 2015-01-27:
I would also like to suggest that you make an organization instead of having all of this on an account.

This allows easier contribution and the possibility for teams and hosting keyboard/converter projects all in one public space instead of having everything scattered across different accounts/users (although still being possible for those who want)

BenBergman, 2015-01-27:
An organization with a main repo for the core code and separate repos that reference the core (perhaps through subtrees or submodules) would be a great way to organize all the related projects. I think you can grant different permissions to different repos in an org, so you can give people access to maintain their own keyboard projects without letting them touch the core, if you wanted to do something like that.

p3lim, 2015-01-27:
> I think you can grant different permissions to different repos in an org, so you can give people access to maintain their own keyboard projects without letting them touch the core

That is correct, and exactly what I suggested :smile:

xauser, 2015-01-27:
I wouldn't split up the tmk project in any kind of way. Having all drivers in one project is very convenient for me. Of course, no single person can test all incoming patches alone because of the required hardware, that's why I would assign a maintainer to each keyboard project/subtree. Patches to those projects/subtrees would go from maintainer to tmk project only.

kejadlen, 2015-01-27:
A separate org with one main project with submodules for individual drivers makes sense to me.

nemith, 2015-01-27:
A single repository can be beneficial as well.  For example code refactoring means a single commit to touch every consumer of a library call vs a commit for each sub repository.

p3lim, 2015-01-27:
> A single repository can be beneficial as well. For example code refactoring means a single commit to touch every consumer of a library call vs a commit for each sub repository.

Which means one person (hasu) would essentially have to either update all of the projects himself or validate and merge changes.  
If we have a main project treated as a submodule it won't be updated unless the repository owner/maintainer of the keyboard/converter repository does so, which also means that nothing will be broken if the main project changes, and leaves the owner/maintainer to update it if necessary.

BenBergman, 2015-01-27:
@p3lim ++

p3lim, 2015-01-27:
@tmk If you end up with the organization path and would want to use "tmk" for the name but still keep all your commits correct, I'd suggest first renaming your account to "hasu" (or anything for that matter, something that would be more personal), then create a new org with the name "tmk", finally transfering the repositories and restructuring everything in the org.
This way you would still be the correct owner/commiter of all the commits, GitHub manages everything on their end.

xauser, 2015-01-27:
I still don't like the idea of having submodules. You split up code that belongs together. You can always create integrity by branches, tags and test suites.

If you refactor core code you have to deal with all your dependencies, that's life. And if you can't test a patch yourself because of time or hardware, let others do it for you.

p3lim, 2015-01-27:
> You split up code that belongs together. You can always create integrity by branches, tags and test suites.

Sure, it belongs together, but in reality the core is treated like a library, and having the _actual_ projects being subdirectories within is (in my opinion at least, seems like hasu would agree considering this issue) cluttered and.. weird.

BenBergman, 2015-01-27:
> I still don't like the idea of having submodules. You split up code that belongs together. You can always create integrity by branches, tags and test suites.

Tags would help a lot for identifying working commits for specific projects, but keeping all the projects in one repository kind of implies that the main branch will be maintaining all projects. Maintenance branches can work, but that is very nearly the same as having a separate repo for each project.

> If you refactor core code you have to deal with all your dependencies, that's life. And if you can't test a patch yourself because of time or hardware, let others do it for you.

That is precisely the point of separated project repos. Worry about the core code and let each project deal with the core changes at their own pace. Using subtrees (instead of submodules), the core code can even stay with the projects so code that belongs together can still stay together for easy access or customization.

domoaligato, 2015-01-29:
I think that it should be as hasu suggested in the OP and structured like cyanogenmod.
have a look here.
http://wiki.cyanogenmod.org/w/Doc:_the_cm_source

tmk, 2015-01-30:
Thanks all for your suggestions and discussion.
To understand them all I needed time to learn about how 'organization' works and difference between submodule and subtree of git.

Orgnization is useful for team development but this tmk_keyboard is still my personal work and keyboard and converter projects(by other developers and including mine) is independent each other in terms of development. I think it is natural and useful that projects are hosted in their own repository and developed in their pace and preference. So I feel like organization is not needed at this moment at least. (that said I don't have much problem to convert my account 'tmk' to organization when its merit becomes clear to me.)

I'd keep 'tmk_keyboard' repository center of tmk firmware information and README(or wiki) of its 'master' branch will have hyper links to third-party project repositories. This is expected to form 'loosely coupled community' of tmk hopefully and I prefer it to github 'organization'. Related project owners can just let me know to add link on the README.

And I'm planing to offer core library(repository or branch?) to be refered hopefully from projects with subtree or submodule. The core library will be what just removed keybaord/ and converter/ directories from current tmk_keyboard repository, instead example project codes may be included perhaps.

In most cases project will have repository specific to a keyboard or converter but you can host several projects in one repository if you maintain many projects like xauser and I do. 'One repository per project' is very clean and useful for (pure) users but may not for developer. In the end it is your repository you can do anything you like! 'tmk' won't mind :) I'll also do what I want.

tmk, 2015-01-30:
Problem is I frequently change core library and break build of some projects and people lose their time to build a broken project report it. And I also have to take time to fix it.

I think 'separaton of core library and projects code' resolves this problem. All projects don't necessarirly need to follow the latest core library, most of projects can stick to old stable commit of the library. So I can leave code of projects intact regardless of frequent change of core library. Now I can change core code without hesitation.

Now what to be considered is how projects already contributed are handled.

p3lim, 2015-01-30:
You could perhaps have all work done in a different branch, then when you find that things are stable you can merge that to master and [tag](http://git-scm.com/book/en/v2/Git-Basics-Tagging) it with a version.
Then the drivers simply specify which version they were developed for.

bjd-pfq, 2015-01-30:
I hereby offer to test, from common usage, and special features upon request, Sun Type 5C keyboards under Linux.

bgould, 2015-01-31:
@tmk I think that is a great decision.  It will be much easier to keep organized.

I can maintain the NeXT converter project if that is helpful to you... I have some improvements to add to the matrix code that I originally contributed... as it happens I use my NeXT keyboard almost every day so I've become very familiar with it.

If I might make a suggestion, if you're not already considering this, it might be a good idea to restructure things so that other projects can just clone the tmk_keyboard core as a submodule of their project which would keep them pinned to a specific version until they choose to upgrade if they want to

That is essentially how I did for my https://github.com/bgould/arduino_tmk_keyboard project and it worked fine.

It might helpful for spinning off projects if the build system could be structured to support that sort of directory layout.  I'd be glad to help test things out once you get started, if you need it.

technomancy, 2015-02-08:
This sounds good to me. I added support for the Atreus in my personal fork a while ago and never got around to submitting it upstream. Now I've found a few things have broken. So I need to keep my Atreus-specific stuff either in a personal fork of this repository or in my own repository that brings this in as a library; the latter is what I prefer even though git submodules kinda suck.

BenBergman, 2015-02-09:
@technomancy Check out git sub-trees. They are a bit more complicated to set up and update, but they solve a lot of the problems of submodules. You can keep the sub repo code in your project and make your own modifications while still keeping up with the upstream code.

aaronz8, 2015-02-13:
I think https://github.com/clibs/clib, a package manager, might help with a lot of the issues brought up here.  I've never used clib before, but it seems similar to https://github.com/npm/npm (node.js package manager) which I have a lot of experience with.

> Organization is useful for team development but this tmk_keyboard is still my personal work and keyboard and converter projects(by other developers and including mine) is independent each other in terms of development. 

I think Github organizations are useful for any collection of related code, not necessarily for team development. It's like having all the related projects in one easy to find place. Not really necessary if you use a package manager unless you're planning to break down the core lib further.

> I'd keep 'tmk_keyboard' repository center of tmk firmware information and README(or wiki) of its 'master' branch will have hyper links to third-party project repositories. This is expected to form [...]

clib comes with a `search` function, which you can use in conjunction with namespacing to easily find all 3rd party tmk projects. For example, for `browserify` in npm: https://www.npmjs.com/search?q=browserify . You could set up a naming scheme like tmk-abc (e.g. tmk-gh60, tmk-nerd) and it'd organize itself.

> And I'm planing to offer core library(repository or branch?) to be refered hopefully from projects with subtree or submodule. The core library will be what just removed keybaord/ and converter/ directories from current tmk_keyboard repository, instead example project codes may be included perhaps.

Sounds like a good example of a module dependency. Keyboard/converter projects would just have project specific code, with the `package.json` file listing core tmk library as a dependency.

> In most cases project will have repository specific to a keyboard or converter but you can host several projects in one repository if you maintain many projects like xauser and I do. 

Considering different projects may have different tmk core version requirements, I would advise against this with or without a package manager. 1 project per repo would simplify dependencies significantly, and github issues would be clearly separated by project.

> Problem is I frequently change core library and break build of some projects and people lose their time to build a broken project report it. And I also have to take time to fix it.

This would not happen with a package manager, since each project must specify dependency versions. You can use git-tag to freeze each version of the core lib, and if a project is stable with that version, it's not forced to change.

p3lim, 2015-02-13:
> > In most cases project will have repository specific to a keyboard or converter but you can host several projects in one repository if you maintain many projects like xauser and I do.

> Considering different projects may have different tmk core version requirements, I would advise against this with or without a package manager. 1 project per repo would simplify dependencies significantly, and github issues would be clearly separated by project.

What I think he meant was that some projects, like the Nerd keyboard, have different sizes, or like the GH60 which have several layouts, those may be part of the same project.

technomancy, 2015-02-13:
> This would not happen with a package manager, since each project must specify dependency versions. You can use git-tag to freeze each version of the core lib, and if a project is stable with that version, it's not forced to change.

You don't need a package manager to solve this; it's easy to avoid using
git submodules. A package manager only makes sense if there's a
nontrivial dependency graph. Maybe if we want to move the LUFA stuff out
of the repo it would be a good fit, but certainly not for the problem
currently under discussion.

p3lim, 2015-03-06:
Got my keyboard running and thought I'd demonstrate a possibility of what we've discussed here.

I added tmk_keyboard as a submodule (subtrees are not easily done on Windows, requires a plugin) and slightly modified the Makefile to support this.

https://github.com/p3lim/keyboard_firmware

xauser, 2015-04-05:
Are there any decisions done now on this topic? I have patches to several projects and don't know how to proceed?

BenBergman, 2015-04-06:
I changed [my project](https://github.com/BenBergman/bluetooth_kinesis) from being a fork to using git subtrees. Nothing really needs to change in the tmk_keyboard repo to allow other projects to take advantage.

I'm not sure if any progress has been made with the existing keyboard projects.

p3lim, 2015-04-06:
@BenBergman You do however need to modify `TOP_DIR` in the Makefile when you're using submodules or subtrees. Other than that, yes, tmk_firmware doesn't need a single change for it to work, but the whole point of this was not only to allow drivers to use tmk_keyboard as a library/framework, but also the separation of the core and the drivers already present in the repository.

tmk, 2015-04-07:
Sorry for late move.
Yes, I think what Ben and p3lim did on their projects is exactly what I want.

I'm going to create new repository(or branch if possible?) whose name is like `tmk_core` so that project owner can submodule or subtree it without including unneeded projects and files(tmk_keyboard/keyboard/* and tmk_keyboard/converter/*).

`tmk_keyboard` will be still main repository for my projects(keyboards and converters) with subtreeing core library codes from`tmk_core`.

I'll look into and try this today.

p3lim, 2015-04-07:
@tmk Branches are also possible if you want to start off easy, both submodules and subtrees support that.

tmk, 2015-04-09:
Created `core` branch which can be used as library with `subtree` or `submodule`. If branch is not useful for this purpose repository(named like `tmk_core`) can be created.
https://github.com/tmk/tmk_keyboard/tree/core

And `master_ng` branch contains `core` library as `subtree` and my keyboard and converter projects. `master_ng` is going to take palce of current `master` branch later if it seems to works.
https://github.com/tmk/tmk_keyboard/tree/master_ng

Both branches are not cleaned and fixed enough yet but I think the idea itself is clear to you.

For example, to import `core` library into  project repository
```
$ cd your_project
...
$ git subtree add -P tmk_core --squash  https://github.com/tmk/tmk_keyboard.git core
or
$ git submodlue add -b core https://github.com/tmk/tmk_keyboard.git tmk_core

$ tree
README
Makefile
src
├── matrix.c
├── keymap.c
└── ...
tmk_core
├── common
├── common.mk
├── doc
├── ldscript_keymap_avr5.x
├── protocol
├── protocol.mk
├── README.md
├── rules.mk
└── tool
```

Am I missing something under discussion so far?

p3lim, 2015-04-09:
If you end up going this path, I'd suggest you split up the repository to avoid a branch mess and confusion because of branches (not everyone is familiar with Git), having `tmk_core` (or whatever you'd prefer to name it) it's own repo, separate from the drivers you'll maintain yourself.

A further improvement upon that would be splitting drivers into their own repositories, preferably in an organization, to allow other maintainers to host all their drivers in a collaborative space, and allow for a cleaner commit log and easier merging. But that is entirely up to you.

tylert, 2015-08-26:
I'm only just starting out on my mechanical keyboard journey now.  Thank you to @tmk and all contributors for making this journey so fun and easy by providing such an excellent open-source project.

Now, as for the issue at hand, I found it very helpful to be able to use the example keyboards provided under the "keyboard" directory as templates that gave me some source files that were "close enough" for my own project.  It was also very helpful to be able to find a working example of a stand-alone project, such as the one provided by @p3lim, showing one way to make use of tmk_core for new keyboard projects that have a low chance of ever being accepted back into the master branch (and to comply with the goals of #173 to keep stuff seperate).

That being said, it would perhaps still be worthwhile to try to encourage folks to issue pull-requests and keep track of such "external" projects with links in the main README for *some* of the clones out there in the wild as was done for the ErgoDox and mcdox ones.  This will continue to help others in the future to find other good reference projects out there to help them get started.

As for the comments earlier about github organizations, one advantage of creating an organization is that it can improve the project's "bus factor" (https://en.wikipedia.org/wiki/Bus_factor).  It does not preclude *also* having your own clone "USER/tmk_keyboard" be a fork of "ORG/tmk_keyboard" too;  You could just set up 2 different remotes in your local clone to allow you to push changes directly to both repos--say, "origin" for the main org's project and "personal" for your own fork.

I also noticed that https://github.com/tmk/tmk_core is still an empty repo too even though it was created over 4 months ago.  One could make use of the 2 remotes trick mentioned above to keep the tmk_core repo in sync with the tmk_keyboard "core" branch.

p3lim, 2016-01-16:
I'm going to bring this topic back up again.

@tmk, when you make changes to the core you seem to do the changes on `master`, then push the changes upstream to the `core` branch. If I were you I'd do it the other way around, making `core` the development branch.

The reason behind this is because of several points I've brought up in this discussion; when you make major changes you can, and will, break drivers.  
Also, when you make changes and forget to push upstream to `core` (which is the case right now), any changes you've done is not available to those who use that branch in other projects/repositories.

p3lim, 2016-01-16:
Also, I just noticed the [tmk_core](https://github.com/tmk/tmk_core) repository, which is not equal to the `core` branch in this project, just adding more confusion.

tmk, 2016-01-16:
I've tried both branch and repository for tmk_core for a while and I feel like repository is easier to work with. I'll remove core branch some later probably.

p3lim, 2016-01-16:
If you indeed intend to use the [tmk_core](https://github.com/tmk/tmk_core) repository as the source for the core, please start using that as your base of development.

Currently the [core repository](https://github.com/tmk/tmk_core) (and [branch](https://github.com/tmk/tmk_keyboard/tree/core) in this repository) is 2 months and 25 or so commits behind the [tmk_core directory](https://github.com/tmk/tmk_keyboard/tree/master/tmk_core) in this repository (which is practically impossible to use in other projects).

p3lim, 2016-01-16:
I just noticed you've made a wiki page for how you handle subtrees and the core:
https://github.com/tmk/tmk_keyboard/wiki/Git-subtree

Development on the core in a separate repository simplifies this a lot, and it also allow people to easily pull the changes too.

---
#### Setup
One-time setup for the subtree in the driver repository (*tmk_keyboard* in your case).

Add a new remote URL for the core repository
`git remote add -f core https://github.com/tmk/tmk_core`

Add the core as a subtree in *tmk_keyboard*:
`git subtree add -P tmk_core core master --squash`

---
#### Updating the subtree
After you've made changes to the core (*tmk_core* repository) you can pull the changes like this:
`git subtree pull -P tmk_core core master --squash`

xauser, 2016-01-16:
How are patches to the keyboard section of tmk_keyboard handled? I have patches for nerd, lightsaber and others in my tree. Will they be maintained in tmk_keyboard repo in future or removed from there? Users should be able to find the latest sources easy.

p3lim, 2016-01-16:
> How are patches to the keyboard section of tmk_keyboard handled? I have patches for nerd, lightsaber and others in my tree. Will they be maintained in tmk_keyboard repo in future or removed from there? Users should be able to find the latest sources easy. 

Anything related to the drivers (anything within the *keyboard* and *converter* directories of this repository) are still being maintained in this repository.
Hasu might change his mind later and separate them into different repositories, that's all up to him.

If it were up to me I'd make an organization and split the drivers into their own repositories.

tmk, 2016-01-16:
@xauser, you can send PR to original author's repository or keep your own repository up to date so that people can clone. People can find the repository from this wikipage. (I think you are author of those projects, though.)  https://github.com/tmk/tmk_keyboard/wiki/TMK-Based-Projects
I will add link to that page in tmk_keyboard/README.md so that people can find the repositories easily.

This is my intention here and I hope it works.

tmk, 2016-01-16:
@p3lim, thanks for subtree instruction. I don't get used to usage of subtree and can't feel comfortable wit it yet, to be honest. It will take time for me to learn new thing as always.

Yes, I want people to subtree(or submodlue) tmk_core repository instead of tmk_keyboard when they start their own keyboard project(driver) repository. They can add their repository URL on the wikipage.

tmk, 2016-06-22:
Moved some projects to `orphan` directory now. 649151cbeb84

vovcacik, 2017-07-08:
I am sorry but this is confusing. You are developing in [tmk_core directory][core_dir], but you want people to use the [tmk_core repository][core_repo] which seems to be outdated. The [tmk_core repo][core_repo] also states the the source code is to be taken from the [core branch][core_branch] which has been deleted.

[core_dir]: https://github.com/tmk/tmk_keyboard/tree/master/tmk_core
[core_repo]: https://github.com/tmk/tmk_core
[core_branch]: https://github.com/tmk/tmk_keyboard/tree/core

tmk, 2017-07-08:
You can think tmk_repo is stable because it is tested here :p I'll update tmk_core repo occasionally.
to be honest, git subtree is cumbersome to me and it prevents me from syncing the repo. I'm thnking about using submodule for tmk_core in this repo. submodule is probably easy to maintain.

p3lim, 2017-07-08:
They are pretty much equally easy to maintain, only major difference is that submodule doesn't pollute the commit log.

tmk, 2017-07-08:
I don't think so with my current knowledge. but I'll learn something anyway after trying submodule.

p3lim, 2017-07-09:
The upside with subtrees is that they are transparent to the user, as in, users don't need to do anything extra.
With submodules they need to clone with an additional parameter (`git clone URL --recursive`).

p3lim, 2017-07-09:
Here's a short guide to submodules like the one I did for subtrees.
---

#### Setup

One-time setup for the submodule in the driver repository (_tmk_keyboard_ in your case).

Add a new submodule on the path "tmk_core":
`git submodule add https://github.com/tmk/tmk_core tmk_core`

Initialize and update the submodule:
`git submodule update --init`

---

#### Updating the subtree

After you've made changes to the core (_tmk_core_ repository) you can pull the changes like this:
```
cd tmk_core
git fetch
git checkout master
```

Here is a great (and complete) write-up:  
<https://medium.com/@porteneuve/mastering-git-submodules-34c65e940407>

rhaberkorn, 2021-02-08:
I'm about to release my own controller based on tmk - it's for the VEB Robotron K7637 keyboard in case you wonder.
I'm unsure how to proceed. The best way would certainly be to add [tmk_core](https://github.com/tmk/tmk_core) as a submodule to a repository containing only my new controller firmware. But the tmk_core repository seems to be hopelessly outdated.
So I will fork [tmk_keyboard](https://github.com/tmk/tmk_keyboard) and do my work on a dedicated branch that I can rebase if necessary. This seems to be what most people do.

@tmk Why don't you use a tmk_core submodule in the development of tmk_keyboard? I would suggest to
1. throw away all the history of the tmk_core repo:
* rename the existing master branch, so the Git commits will stay around and you don't break anybody's repo
* create a new completely empty master branch
* pull from tmk_keyboard, effectively making tmk_core a clone of tmk_keyboard
* remove any controller and converter and move the contents of the tmk_core subdirectory to the root of the tmk_core repo - this will make sure that history of tmk_core from tmk_keyboard is preserved
2. add the newly synced tmk_core repo as a submodule to tmk_keyboard

tmk, 2021-02-09:
I still have a plan and  will do that in the end. But it won't happen some time soon.
I don't have enough time to do properly and another reason is my laziness.

tmk_core was updated using git-subtree merge and this makes process a bit complex.
I'll base on submodule instead next time.

tmk, 2021-02-11:
The tmk_core repo was updated with change sets from this repo now.
https://github.com/tmk/tmk_core/commit/8bfaa6ce5380323c77427cf2a01ae2092db5ff49

This repo will be changed later once my topic branches in progress are settled.

rhaberkorn, 2021-02-13:
I have published my project using a tmk_core submodule as was suggested.
Could you add a link to my project, please?

[VEB Robotron K7637 Replacement Controller and Firmware](https://github.com/rhaberkorn/tmk7637)

tmk, 2021-02-13:
Interesting project! Done.