tmk/tmk_keyboard #179: bootloader_jump doesn't work

kejadlen, 2015-01-26:
I feel like I must be doing something wrong, but I can't get `bootloader_jump` to work. Is there anything else I need to do besides changing `BOOTLOADER_SIZE` in `config.h` and using the following code in the keymap?

``` C
if (id == TEENSY_KEY) {
    clear_keyboard();
    print("\n\nJump to bootloader... ");
    _delay_ms(250);
    bootloader_jump(); // should not return
    print("not supported.\n");
}
```

Source here, if that helps at all: https://github.com/kejadlen/tmk_keyboard/tree/master/keyboard/planckeus

I also inadvertently ran the bootloader jump with the wrong bootloader size before realizing that needed to be changed - would that have any effect on future uses of `bootloader_jump`?

tmk, 2015-01-26:
I glanced over your codes; keymap and Makefile. It looks to me ok and should work.

And jummping wrong address won't have any permanent effect on hardware or firmware.

kejadlen, 2015-01-26:
Ah, I didn't notice this in the output before:

```
keymap_default.c: In function 'action_function':
keymap_default.c:39:9: warning: implicit declaration of function 'bootloader_jump' [-Wimplicit-function-declaration]
         bootloader_jump(); // should not return
         ^
```

Could this be why it's not working?

kejadlen, 2015-01-26:
Interestingly, the physical button doesn't work after I try to get into the bootloader via the action function. I need to unplug the USB cable, re-plug it back in, and then hit the physical button for it to do anything.

tmk, 2015-01-27:
Weird, reset button should work all the time even if firmware goes mad.
I don't have Teensy to test at hand and don't know if bootloader_jump code really works for Teensy.

tmk, 2015-01-27:
You can try this old code instead of current bootloader_jump(). 
https://github.com/kejadlen/tmk_keyboard/blob/master/common/avr/bootloader.c#L87

The old code is from PJRC article and expected to work with Teensy bootloader.
https://www.pjrc.com/teensy/jump_to_bootloader.html

kejadlen, 2015-01-27:
Drat, forgot to mention that `bootloader_jump` works just fine on my Ergodox: https://github.com/kejadlen/tmk_keyboard/tree/master/keyboard/ergodox, so it seems the current code does at least work for the Teensy there.

technomancy, 2015-02-08:
I can confirm this on my Teensy: https://github.com/technomancy/tmk_keyboard/commit/876f356900504a8d38af72c918c1373073df0cf3

kejadlen, 2015-02-08:
@technomancy Does that SHIFT macro work? I'd love to get rid of the tons of `fn_actions` I have for shifted keys at the moment.

technomancy, 2015-02-10:
It doesn't work, but I have a lead on how to make it work; see #184.

technomancy, 2015-02-12:
@kejadlen `bootloader_jump` still isn't working for me, but I got the `SHIFT` macro working on my `teensy2` branch.

p3lim, 2015-03-06:
Add this to the top of the file where you're using `bootloader_jump()`:

```C
#include "bootloader.h"
```

kejadlen, 2015-03-06:
@p3lim Tried that, but it doesn't work for me.

p3lim, 2015-03-06:
@kejadlen Any errors when compiling?

kejadlen, 2015-03-06:
@p3lim No, it compiles just fine, but trying to use `FN0` (the bootloader jump) freezes/crashes the keyboard so I need to reset it by pulling the USB plug.

tmk, 2015-03-06:
What's your keyboard hardware in fact? and Which Teensy?
You said you can't use reset button when it freezes. Doesn't pushing reset button for long like 5-10 sec work for you?

Seems like it jumps wrong address and goes mad.

kejadlen, 2015-03-06:
Using a Teensy 2.0. Pushing it for 5-10 seconds doesn't do anything either. 

That sounds plausible to me. Still not sure what the difference is between what I'm doing and other Teensy 2.0 projects, though. ([source](https://github.com/kejadlen/tmk_keyboard/tree/master/keyboard/planckeus))

technomancy, 2015-03-12:
@kejadlen I got this working by just dumping in some code from pjrc.com: https://github.com/technomancy/tmk_keyboard/commit/ebb4f96b074e06be8ab3ee3727ad384f5b78f58b

Probably would be better if it were fixed properly inside the TMK common code, but this works for me.

tmk, 2015-03-13:
Hmm, it smells like this line is fishy. Perhaps it can be compiled differently depending on compiler version?

https://github.com/tmk/tmk_keyboard/blob/master/common/avr/bootloader.c#L75

We should use assembly code fragment there?

kejadlen, 2015-03-13:
Curiously, `bootloader_jump` works correctly on the Ergodox, which uses a Teensy 2.0 as well. I have no clue what the difference might be.

p3lim, 2015-03-13:
Can confirm it works fine on a Teensy 2.0, here is how I've added it:
https://github.com/p3lim/keyboard_firmware/blob/d3d65d8de7a6a3e876907466cdba44e3f9ec3c84/keymap.c#L88-L99

Do note that this requires the following includes, as I have in the top of the file:
```C
#include "bootloader.h"
#include "command.h"
```

kejadlen, 2015-03-14:
@technomancy Thanks, that workaround works for me as well.

Unfortunately, I'm fairly out of my depth here, but I'd be happy to investigate further if anyone has any suggestions for what I could do to provide more info.

tmk, 2015-04-13:
I seems to me like GCC changed bahaviour somewhere in 4.8 or 4.7.

This C code is comipled with `icall` which requires ***word address*** in GCC 4.8.1. Probably older compiler spits out assembly mnemonic requries ***byte address***, though I confirmed yet.

```
 ((void (*)(void))(BOOTLOADER_START))();
```

I think using inline assembler is safer again future compiler behaviour changes, like PJRC describes here. https://www.pjrc.com/teensy/jump_to_bootloader.html
```
asm volatile("jmp 0x7000");
```

Old LUFA document expects GCC uses byte address to jump.
http://www.fourwalledcubicle.com/files/LUFA/Doc/120730/html/_page__software_bootloader_start.html

New documents make use of word address with logical shift(>>1).
http://www.fourwalledcubicle.com/files/LUFA/Doc/140928/html/_page__software_bootloader_start.html

Similar icall and byte/word address problem:
http://www.avrfreaks.net/forum/dfu-bootloader-immediately-bails-out

tmk, 2015-04-13:
TMK had refered to old wrong LUFA documentaion. Jump address should be word counted.
http://www.fourwalledcubicle.com/files/LUFA/Doc/120219/html/_page__software_bootloader_start.html

LUFA fixed it.
http://www.fourwalledcubicle.com/files/LUFA/Doc/140928/html/_page__software_bootloader_start.html
https://github.com/abcminiuser/lufa/commit/c5282637edde46f2d5ecec62d987ede0f543e889

GCC version is not related at all.

tmk, 2015-04-23:
TMK fixed this problem at Sep 18, 2013 with this commit 0ca415004a453b2a841880d3a66492c664505737.

Make sure setting proper value to `BOOT_LOADER_SIZE` in Makefile.
```
# Boot Section Size in *bytes*
#   Teensy halfKay   512
#   Teensy++ halfKay 1024
#   Atmel DFU loader 4096       (TMK Alt Controller)
#   LUFA bootloader  4096
#   USBaspLoader     2048
OPT_DEFS += -DBOOTLOADER_SIZE=4096
```

closed 2015-04-23 by tmk

AxisRay, 2015-10-03:
I have the same problem.
It seems that GCC version is related.
I have a Teensy 2.0 ++ with chip AT90USB1286,and compile tmk with WinAVR (gcc-avr 4.3.3)
I should use
> ((void (*)(void))(BOOTLOADER_START))();
instead of
> ((void (*)(void))(BOOTLOADER_START/2))();//It doesn't work
but when I use gcc-avr 4.8.1 under ubuntu ,both of them don't work.
so I use the following code instead
> asm volatile("jmp 0xFC00");//I don't know why not 0x1FC00

tmk, 2015-10-07:
hmm, I have to check WinAVR again some time later.(TODO)

`BOOTLOADER_SIZE` for Teensy++2.0 is 2048 bytes, though some of my Makefiles say it is 1024 wrongly. Did you use correct size?

I confirmed it works on Teensy++2.0 with gcc 4.8.1/avr libc1.8.0.

AxisRay, 2015-10-08:
Thanks for reply.
I have checked BOOTLOADER_SIZE,but It still doesn't work.(test with gcc 4.8.1/avr libc1.8.0.)
Well,maybe there is something wrong with my teensy.
I have no patient to figure out what going wrong with it,the inline assembler works well after all.
Anyway,Thanks for your firmware.It very useful for me.

tmk, 2015-10-08:
Thanks.
You still have problem with GCC 4.8.1, this clearly indicate this is not related to GCC version.

I think WinAVR is not guilty.