tmk/tmk_keyboard #180: Possible to hack for more than 32 fn_actions

mkcode, 2015-01-28:
Does anyone know if it is possible for me to have more than 32 fn_actions?

I saw @cub-uanic do it here: https://github.com/cub-uanic/tmk_keyboard/blob/master/keyboard/ergodox/keymap_cub.h#L567-L595
but that implementation is dependent on layers, I'd like to have more than 32 in one layer.

Reason for this is I'd like to have all shifted keys available without using a shift - so I am using 
`[30] = ACTION_MODS_KEY(MOD_LSFT, KC_9)` to get access to `open parens`. After adding all of these keys and with my other dual-role action functions, I am out of space.

tmk, 2015-01-30:
no easy fix, restriction of 32 comes from 8-bit keycode space. As you can see no unused keycodes left.
https://github.com/tmk/tmk_keyboard/blob/master/common/keycode.h

You may be able to use action codes instead of keycodes. Internally keycodes is subset of action codes and action codes have no limitation of number of FN action. You can define direclty actions in actionmap like below. But I've never used actionmap myself in fact.

```C
keymap[][][] = {
    KEYMAP(FN1, Q, ....
}

actionmap[][][] = {
    { ACTION_MODS_KEY(MOD_LSFT, KC_9), ACTION_KEY(KC_Q), ...
};
```

mkcode, 2015-02-01:
Thanks for the info tmk. I will try the action maps.

arvids, 2015-02-02:
@mkcode Did it work?

closed 2015-07-12 by tmk

pogues, 2018-01-26:
a bit late to the party but for reference using the unimap (action codes) works.  I could not get the actionmap style noted by tmk running after a couple of hours messing around.
I have committed what I did here: https://github.com/pogues/tmk_keyboard/blob/master/keyboard/hhkb/unimap_vim.c if it is of use to anyone.

tmk, 2018-01-27:
Unimap is basically actionmap which is used for TMK products to offer web keymap editor.
I think you can use unimap without problem.

you can buid and flush with commands below:
```
make -f Makefile.unimap KEYMAP=vim clean
make -f Makefile.unimap KEYMAP=vim
make -f Makefile.unimap KEYMAP=vim dfu
```

pogues, 2018-01-27:
Yes - that is exactly how I built the version I referenced - which is working well and has about 40 items in it that would previously have been fn_actions entries.  Over time I can't stop tweaking so I think there will be more added almost daily!

I got confused by the code snippet and had tried to create an actionmap directly (not using the UNIMAP_HHKB macros provided ) and ended up chasing my tail - hence deciding to link the version I finally created the simpler way here in case anyone else did the same thing and just wanted a starting reference (or maybe they are all smarter than me :-))

tmk, 2018-01-27:
Ah, I see. I'll have to add what is actionmap/unimap and how to define somewhere, probably wiki.

Thanks for the code, it would be helpful for someone intrested.