tmk/tmk_keyboard #184: `ACTION_MODS_KEY` does not send modifier.

technomancy, 2015-02-08:
I'm trying to use `ACTION_MODS_KEY` to send shifted keycodes. In `keymap.md` it says you can use `ACTION_MODS_KEY(MOD_LSFT, KC_1)` to send a shifted `1`, or `!`. However, when I bind this to a key inside the `keymaps[]` array, it just sends `1`.

```c
  KEYMAP(SHIFT(KC_1), SHIFT(KC_2), SHIFT(KC_MINS), SHIFT(KC_EQUAL), SHIFT(KC_BSLS), ...)
```

I have changed the `KEYMAP` macro to use a 16-bit value instead of an 8-bit one as the documentation indicates is necessary.

I have this working using `ACTION_MODS_KEY` in the `fn_actions` array, but it's far too tedious. I need to be able to declare it inline in the keymap since I have layers that are mostly shifted keycodes. Having to declare every shifted keycode once in the keymap and once in `fn_actions` (and trying to keep the references straight when I change them) is error-prone.

Is this supposed to be supported in `KEYMAP` or only inside `fn_actions`?

tmk, 2015-02-08:
To use action directly like that you will have to implemnt `action_t action_for_key(uint8_t layer, keypos_t key)` youself  Intead of using keymap.c. The function should convert key event to action, *keymap.c* uses user defined keymaps to convert. Your version of the function should use actionmaps(16bit map) instead of 8bit keymap.

technomancy, 2015-02-12:
Thanks; I've got this working in 2f7b588, but not cleanly; I just ripped out a bunch of code I'm not using to make it function.

I would love it if this were possible to do in the mainline TMK codebase and was considering submitting a pull request with this done the Right Way. But I get the feeling that there's a reason it's not supported currently. Is it just a reluctance to use 16 bits in order to make the layer definitions more compact in memory?

ardumont, 2015-02-22:
@technomancy @kejadlen I can confirm it also works for `CTRL` macro - https://github.com/ardumont/tmk_keyboard/blob/qwerty-modified-with-ctrl-i-and-ctrl-m/keyboard/atreus/keymap_qwerty_modified.c#L22

closed 2016-03-10 by tmk