tmk/tmk_keyboard #186: Documentation for keymap_common?

aaronz8, 2015-02-11:
I've been following the tutorial at http://deskthority.net/workshop-f7/how-to-build-your-very-own-keyboard-firmware-t7177.html to try to use tmk with FaceW PCB. However, I am stuck at http://deskthority.net/workshop-f7/how-to-build-your-very-own-keyboard-firmware-t7177.html#p141386, where it skips over the keymap_common.h configuration. I tried looking for other examples, such as 
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/nerd/keymap_common.h, but I'm still confused where these numbers come from.

Is there documentation for this anywhere?

image of the FaceW row,column connections for each key:
[image]

aaronz8, 2015-02-11:
I've figured it out. I'll do a write up when I've tested everything.

closed 2015-02-11 by aaronz8

Leimi, 2015-04-25:
Hey,

Kinda late to the party but did you end up getting something working/writing down anything on it? I'd love to get my FaceW PCB working with TMK but I'm terrible at all this.

aaronz8, 2015-04-28:
Hi @Leimi,
Sorry for the delayed response. Unfortunately I soon discovered that the FaceW PCB is based around the Atmega32a, which does not have integrated USB and thus not supported by the tmk firmware. The only bootloader I could find supporting the atmega32a was [bootloadhid](https://www.obdev.at/products/vusb/bootloadhid.html), along with its virtual usb implementation, [V-USB] (https://www.obdev.at/products/vusb/index.html). I looked into using V-USB with tmk but gave up once I found out it requires libusb and libusb doesn't work well with HID on Mac (http://www.libusb.org/ticket/89).

Leimi, 2015-04-28:
Hey,

Oh well that's a bummer. Thanks for the feedback!