tmk/tmk_keyboard #187: Feature Request: Reset Teensy Button!

joshfrese, 2015-02-13:
First off, thanks for the firmware. It would be nice to have a way to map a keyboard press to reboot the teensy or put it into the mode that accepts new software.

p3lim, 2015-03-02:
<kbd>LShift</kbd>+<kbd>RShift</kbd>+<kbd>Pause</kbd> will jump to the bootloader.
It's part of the magic commands.

kejadlen, 2015-03-02:
See https://github.com/cub-uanic/tmk_keyboard/blob/master/keyboard/ergodox/keymap_cub.h#L468-L474 for an example of how to use it in a keymap.

p3lim, 2015-03-02:
You can also override magic commands, like so:
```C
bool command_extra(uint8_t code)
{             
    switch (code) {
        case KC_B:
            clear_keyboard();
            print("\n\nJump to bootloader... ");
            _delay_ms(1000);
            bootloader_jump(); // not return
            print("not supported.\n");
            break;
    }
    return false;
}
```

joshfrese, 2015-03-02:
Ah, very nice thanks.

closed 2015-03-02 by joshfrese