tmk/tmk_keyboard #188: Auto-detect USB device path for non-teensy?

technomancy, 2015-02-14:
When using `make teensy` you don't have to specify the path to the USB device since `teensy_loader_cli` does this for you. But when using `avrdude` you need to write out your own `PROGRAM_CMD` option that includes the path to the appropriate USB device, which can be confusing since there's no one cross-platform setting you can use for this.

What if we added a `make avrdude` target that could invoke `avrdude` with a guess the correct auto-detected arguments based on the current OS?

tmk, 2015-03-13:
If it can be implemented easily  avrdude should have that option already. Lastime I skimmed the manual I didn't find it.

technomancy, 2015-03-13:
I figured it couldn't be that hard since the teensy loader does this, but it loops through all USB devices using syscalls and looks for a specific Vendor ID and Product ID match:

  https://github.com/PaulStoffregen/teensy_loader_cli/blob/master/teensy_loader_cli.c#L213

Probably wouldn't work here since the Vendor ID and Product ID can change. Too bad.

closed 2015-03-13 by technomancy