tmk/tmk_keyboard #194: Reassign shift+key symbols

sypl, 2015-02-26:
Is it possible to reassign the shifted symbols, like how some keys of the HHKB2 JP does: http://xahlee.info/kbd/i/happy_hacking_keyboard_jp_layout.jpg

Note how shifted semicolon is a plus and shifted colon is an asterix, as well as a few unorthodox remappings. I don't know how this can be done in tmk. I was looking at 'momentary switching with modifiers', but this only works on my self-defined layer, right? Can the keymap target the shift layer?

sypl, 2015-02-26:
Scratch that, I think the  'momentary switching with modifiers' is fine. I just misunderstood it. Will try this tomorrow.

sypl, 2015-03-02:
I tried but didn't succeed. I used the function to activate a layer, and whilst the shift registered whilst the key was held down, it also meant I was only able to register shifted characters, so " and : were okay, but ' and ; could not be produced. Is there any other way, or is there a trick I'm missing?

kevinkn, 2015-03-02:
This issue has been talked about with the programmer dvorak topic. I dont
think it was solved though.

closed 2015-03-14 by sypl