tmk/tmk_keyboard #195: Media keys on Mac

bhaap, 2015-03-04:
The KC_MNXT and KCMPRV does not work on MAC, The MPLY works thou, have i missed how to create this?

tmk, 2015-03-06:
hm, I don't have Mac nor idea on this.

I think you may be able to know what keycodes you can use on mac by skiming the sources around here.
https://opensource.apple.com/source/IOHIDFamily/IOHIDFamily-606.1.7/IOHIDFamily

tmk, 2015-03-07:
Hm, looks to me like those keys sould work.

`kHIDUsage_Csmr_ScanNextTrack`  and  `kHIDUsage_Csmr_ScanPreviousTrack` handled in `IOHIDConsumer::dispatchConsumerEvent` are `KC_MNXT` and `KC_MPRV` in TMK.

https://opensource.apple.com/source/IOHIDFamily/IOHIDFamily-606.1.7/IOHIDFamily/IOHIDConsumer.cpp

bhaap, 2015-03-08:
When running usb overdrive it recognizes the next track button on my keyboard as a different next track button then on the internal keyboard, wonder if there is 2 almost identical codes for next track. Do you know of any program that can see what scancodes that im sending with the internal keyboard for mac?

tmk, 2015-03-08:
I don't know how to get scancode on OS X. But from reading source codes I guess USB Apple Keyboard sends F6 or F8 as normal keycode and Fn in Apple vendor specific usage page.

What do you mean with "internal keyboard" actually? Macbook builtin keyboard?
And I don't know much about how Mac media keys work. Fn+F6=previous track and Fn+F8=next track, right?

bhaap, 2015-03-08:
Yeah the internal keyboard on my laptop :) f7 and f9 is prev and next. i could be like that

bhaap, 2015-03-08:
But the config in systemprefs only works on the built in keyboard not on my tmk one even thou it has a fn layer

tmk, 2015-03-08:
Fn key on tmk is completely different from Mac. It doesn't send any key code basically. Mac mediakey functions on macbook and apple keyboard are vendor specific unfortunately, systempref works only for apple hardware.

But as I said previously Mac seems to support standard HID media control codes too. Tmk can send only the standard codes.

To me weird thing here is why MPLY only works.

Any other Mac user here? can you confirm this problem?

bhaap, 2015-03-08:
The mnxt and mprv do work with itunes i just noticed, but not with spotify or vlc.

tmk, 2015-03-08:
Interesting.
I'll look into OSX sources again later. It may handle apple media key and standard one as different key events.

bhaap, 2015-03-08:
If you want any help just tell me, im not that good with programming but I can troubleshoot for you if you want to :D

kairyu, 2015-03-08:
Do you have tried this #160 ?

bhaap, 2015-03-08:
I have not will try it

tmk, 2015-03-09:
Thanks, kairyu.
Oops, this issue is exactly same as that. It was not so long ago, I blame my short memory :D 

>The "Play/Pause"(KC_MPLY) works good on Apple OS. But "Next/Prev Track" buttons are unfunctional.
Apple OS use "Fast Forwad"(F9 on Apple keyboard) & "Rewind"(F7 on Apple keyboard) instead.

bhaap, close this issue if it works for you.

tmk, 2015-03-09:
Added FAQ entry.
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#media-control-keys-in-mac-osx

bhaap, 2015-03-09:
The MFFD and MRWD does not work for me. Not in itunes nor in spotify.

tmk, 2015-03-09:
Oh, shame.
We need other Mac users's help then.

bhaap, 2015-03-09:
The MNXT and MPRV do work as i said with itunes, but for some reason it doesnt work with Spotify and other 3rd party apps. Very wierd.

bhaap, 2015-03-11:
Got this from a mac programmer:
"They go through the OS (as in: they trigger events that applications can see), but they aren't delivered as keyboard events. They're system-defined events. So the very concept of key codes doesn't apply to them.

The easiest way to capture them without having iTunes interfere is setting up an event tap for NX_SYSDEFINED/NSSystemDefined events. From there, one way to identify individual media keys is converting the tapped CGEvent to an NSEvent and looking at the NSEvent's -data1."

kyleboyle, 2015-05-09:
I am also having this issue. MNXT and MPRV don't do anything system wide (spotify, vlc etc), and the PR #160 that implements MFFD and MRWD also don't do anything, I would say that PR doesn't solve any issues. I'm happy to help as my ergodox won't be acceptable as my daily driver until I can control media on OSX.

tmk, 2015-05-09:
You need this Fix too.

https://github.com/tmk/tmk_keyboard/pull/199

tmk, 2015-07-03:
https://geekhack.org/index.php?topic=71702.msg1789509#msg1789509
Confirmed.

On Mac just use `KC_MFFD`(`KC_MEDIA_FAST_FORWARD`) and `KC_MRWD`(`KC_MEDIA_REWIND`) instead of KC_ and KC_MNXT and KC_MPRV.

All fixes were already merged in my TMK repository you don't need any patch for this.

For Ergodox version(https://github.com/cub-uanic/tmk_keyboard) you need to apply two patches.
https://github.com/cub-uanic/tmk_keyboard/issues/22

closed 2015-07-03 by tmk

tmk, 2015-11-04:
TODO: 
add description in keymap document

tmk, 2016-01-11:
Added this to FAQ.

https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#media-control-keys-in-mac-osx

unrealhoang, 2016-03-28:
I'm currently using OSX El Capitan (10.11.3), MNXT & MPRV are working correctly but MPLY is not working at all.

tmk, 2016-03-28:
Huh, confusing as hell. Ditch OSX or give up those keys.

kaansoral, 2016-03-28:
I've been using TMK on OSX for months, MPLY worked for me all this time, didn't have any issues at any point

unrealhoang, 2016-03-28:
hmm, my current keymap: 
```
#include "keymap_common.h"

const uint8_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    /* 0: qwerty */
    KEYMAP_ANSI(
        ESC, 1,   2,   3,   4,   5,   6,   7,   8,   9,   0,   MINS,EQL, BSPC, \
        TAB, Q,   W,   E,   R,   T,   Y,   U,   I,   O,   P,   LBRC,RBRC,BSLS, \
        LCTL,A,   S,   D,   F,   G,   H,   J,   K,   L,   SCLN,QUOT,     ENT,  \
        LSFT,Z,   X,   C,   V,   B,   N,   M,   COMM,DOT, SLSH,          RSFT, \
        FN0,LALT,LGUI,          SPC,                     RGUI, FN0, APP, CAPS),
    KEYMAP_ANSI(
        GRV, F1,  F2,  F3,  F4,  F5,  F6,  F7,  F8,  F9,  F10, F11, F12, DEL,  \
        TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,VOLD,VOLU,MUTE, \
        TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,LEFT,DOWN,UP,  RGHT,TRNS,TRNS,     TRNS, \
        TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,MPLY,MPRV,MNXT,          MSEL, \
        TRNS,TRNS,TRNS,          TRNS,                    TRNS,TRNS,TRNS,SLCK),
};
const uint16_t PROGMEM fn_actions[] = {
    /* Poker Layout */
    [0] = ACTION_LAYER_MOMENTARY(1),  // to Fn overlay
};
```

MPRV & MNXT works but not MPLY, am I doing anything wrong?

kaansoral, 2016-03-28:
I can't see anything wrong, it triggers, plays, pauses iTunes, or VLC for me, whichever is active

The only possible difference is that I'm using a 4-5 months old firmware, yet I don't think that would make a difference, correct me If I'm wrong

Nephiel, 2016-03-28:
I'm having the same issue on Linux with a build from yesterday.  
- MPRV and MNXT work fine
- MPLY and MSTP do not even send anything at all (xev reports nothing)  

On my previous build (from January 17th) they were all working.

Nephiel, 2016-03-28:
I've looked into this, it's a regression of #199 from commit 5e9b21d77d71bede3fbcbefd7567af22f3b0d153.  
The enum for media control consumer key codes in `keycode.h` got reordered in that commit. Since `keymap.c` only reads the enum from `AUDIO_MUTE` to `MEDIA_REWIND`, the following keys are ignored:
- `MEDIA_STOP`
- `MEDIA_PLAY_PAUSE`
- `MEDIA_EJECT`
- `MEDIA_SELECT`
- `MAIL`
- `CALCULATOR`
- `MY_COMPUTER`
- `WWW_SEARCH`
- `WWW_HOME`
- `WWW_BACK`
- `WWW_FORWARD`
- `WWW_STOP`
- `WWW_REFRESH`
- `WWW_FAVORITES`  

Possible fixes are: restoring the order of the enum, placing `MEDIA_REWIND` at the end, or reverting keymap.c from #199.

unrealhoang, 2016-03-28:
@Nephiel Awesome, It works now. Thank you for taking time investigating this issue

tmk, 2016-03-28:
aaargh, I broke it at the commit 5e9b21d. OSX is not guilty this time lol.
This commit will fix it.
https://github.com/tmk/tmk_keyboard/commit/24cee117d0abe1e8ba9a136f6908f10c30a11bb9