tmk/tmk_keyboard #196: keyup changes across layers

kejadlen, 2015-03-06:
If a key is a modifier on one layer and a normal key on another, that modifier can get stuck when moving across layers.

For example, if a key is shift on L0 and 'a' on L1 and I hold the key down on L0, change layers, and release it on L1, the keyboard can get stuck in shift.

I'm not sure if retaining the state of the key when pressed down is reasonable/possible/easy, but it would simplify my keymaps if this were the case.

tmk, 2015-03-06:
You're right.  This is limitation(by design) of current keymap system implementation. I may change this in future but not near. And I think it is not so difficul to write your own simple keymap system which just meets your need.

https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#modifierlayer-stuck

kejadlen, 2015-03-06:
Thanks! Is there a way to do `ACTION_MODS_TAP_KEY` with a shifted key?

closed 2015-03-06 by kejadlen

tmk, 2015-03-06:
No. You'll have to write code in acton function.