tmk/tmk_keyboard #201: unneeded delay in momentary layer switch

tmk, 2015-04-13:
Press Fn(Layer switch momentary) then a key on the layer immediately, you will see delay before appearing the key on terminal. Note that if 
The delay seems to be related to `TAPPING_TERM`.

Probably tapping code has a bug.

https://geekhack.org/index.php?topic=56494.msg1715297#msg1715297

boblehest, 2015-04-23:
Hey, I noticed the delay as well some time ago (My LEDs indicate the active layer, so the delay was very apparent), and today it hit me what was causing the delay. It's the tapping delay, as you said, so the issue isn't with the tapping mechanism, but the fact that it's being used wrongly. :p

The fix is rather simple, change this define:

    #define ACTION_LAYER_MOMENTARY(layer)    ACTION_LAYER_ON_OFF(layer)

into:

    #define ACTION_LAYER_MOMENTARY(layer)    ACTION_LAYER_INVERT(layer, ON_BOTH)

tmk, 2015-04-23:
Nice.

And this gaves me a clue of where the problem root was.

Fixed at 1f4a22ee81ec1986d1ca190c78c66231d37b50e2.

closed 2015-04-23 by tmk