tmk/tmk_keyboard #208: Modifier keys (Enter, Backspace, Tab) almost don't work with Alt HHKB controller v1

EDmitry, 2015-04-27:
For some reason modifier keys don't get processed at all until I hit them with other keys (for example hit space repeatedly while pressing 'a' at the same time). All non-modifier keys work fine. I am using regular Makefile to build the firmware since I have HHKB PRO 2.

tmk, 2015-04-29:
Not sure what your problem is, describe it more in the detail.

What did you mean with 'modifier keys'? Modifier keys = Enter, Backspace and Tab?
And what does 'Alt HHKB controleer v1' mean in fact?

EDmitry, 2015-04-30:
Modifier keys that do not work are Enter, Backspace and Tab as I mentioned. The controller is the first version of a non-bluetooth controller from https://geekhack.org/index.php?topic=12047.0, I was in line long time ago to get one. If I just hit say `Enter` nothing would happen. If I hit Enter with some other key fast like: `Enter` `a` `Enter` `a` `Enter` `a` then `Enter` will start being recognized for that sequence.

tmk, 2015-04-30:
OK. I wanted to clarify it because those keys are not modifier key in general terminology.
http://en.wikipedia.org/wiki/Modifier_key

hm, I can't come up	with any clear cause and never hear problem like this so far. Random thoughts...

What is you os?
NKRO can be cause it on Mac, turn off NKRO with pressing Lshift+Rshift+n.

Mate connector firmly again, loose 	connection may cause problem.

Did the controller work before? This is your first try since you received it?

Did you flash you own build firmware and change keymap?  Try the latest default firmware.
Just download firmware without editing keymap from here and flash it.
http://www.tmk-kbd.com/tmk_keyboard/editor/hhkb/index.html

If you use USB hub connect the keyboard directly to PC. Some hub can't provide enough voltage.

EDmitry, 2015-05-06:
I have tried recompiling the firmware with NKRO disabled, it didn't fixed an issue. It definitely works with a stock controller, moreover I have used my own controller for a long time with this keyboard and it worked well (it was me posting oscilloscope photos here https://geekhack.org/index.php?topic=12047.msg1310936#msg1310936). I will try it with another HHKB keyboard of mine. I might debug this to see if anything gets registered at all when just `Enter` is pressed down.