tmk/tmk_keyboard #212: Missing cstddef

kaansoral, 2015-05-05:
../../tmk_core/tool/mbed/mbed-sdk/libraries/mbed/api/platform.h:25:19: fatal error: cstddef: No such file or directory

I've tested at least 5 various arm/gcc sources, brew/port, couldn't move past this issue on a mac

On the other hand, the kiibohd/controller is able to build with most arm libraries

Generally, the build/readme's concentrate on AVR, it would be great to have some ARM instructions too

BlindLemonLipschitz, 2015-05-08:
I'm getting the same(cstddef) error when I run ``` make ``` in the infinity directory
I did:
    ```brew tap PX4/homebrew-px4``` 
   ``` brew update```
    ```brew install gcc-arm-none-eabi-48```
for the arm/gcc sources and I'm on Mac 10.10.3

kaansoral, 2015-05-08:
I also tested gcc-arm-none-eabi / gcc-arm-none-eabi-49 and other brew/port sources, none of them worked, I'm guessing it's a mbed issue, rather than a tmk firmware issue directly, however it's preventing my usage of this firmware, it would be great if we could find a solution

BlindLemonLipschitz, 2015-05-08:
I just created a digitalocean ubuntu 14.04 64 droplet and it built fine on that. So I think you're right about it not being a tmk firmware issue directly. But I would also like to be able to build it on osx.

kaansoral, 2015-05-08:
LOL, why didn't I think of that!

I should also use a virtual machine to take care of the compilation and use an actual machine to deploy the firmware, thanks a lot for the idea

But like you mentioned, it's a lot less practical then using a normal machine

I will probably give this a direct shot before going the virtual root: https://launchpad.net/gcc-arm-embedded
(I found a brew formula that handles the installing procedures, however I guess it didn't work, I don't recall the reason tho)

kaansoral, 2015-05-09:
Tested almost all launchpad binaries manually, the brew formulas also just integrate them, it's as simple as downloading them and adding them to path, they are pre-compiled, all of them have the same cstddef issue on osx

tmk, 2015-05-09:
I also had similar problem with GCC4.8 on Ubuntu 14.04.
I had to update to 4.9 with this PPA.
https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded

BlindLemonLipschitz, 2015-05-09:
@tmk 
this worked on my **Ubuntu 14.04 droplet**:
```apt-get update```
```apt-get install git```
```git clone https://github.com/tmk/tmk_keyboard.git```
```cd tmk_keyboard/keyboard/infinity/ ```
```apt-get install make```
```apt-get remove binutils-arm-none-eabi gcc-arm-none-eabi```
```add-apt-repository ppa:terry.guo/gcc-arm-embedded```
__```apt-get install gcc-arm-none-eabi=4.9.3.2015q1-0trusty13```__
```make clean```
```make```
   ```
@kaansoral 
I also tried gcc-arm-none-eabi-49 from PX4/homebrew-px4 with no luck on **OS X**

U47, 2015-08-07:
So what exactly in terry.guo's PPA is different than the PX4 one? Incredibly frustrating that there is no way to build firmwares on a Mac atm.

tmk, 2015-08-21:
On Mac GCC 4.9 and any versions are not solution.
https://geekhack.org/index.php?topic=41989.msg1841940#msg1841940

After some google searches I got my next random guess :)
From my understandings GCC provides 'libstdc++' as standard library while LLVM clang is developing 'libc++'. TMK can be compiled normally with GCC and libstdc++ on Linux. I guess GCC uses libc++ unintentionally for some reason on Mac and TMK/mbed code are not compatible to libc++.

Do you have libc++ somewhere on your Mac? What if it is uninstalled?
I think Homebrew or Xcode may install the library.

I don't have certain knowledge of GCC and C++, this may be completely wrong.

https://gcc.gnu.org/libstdc++/
http://libcxx.llvm.org/
https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/C%2B%2B-Standard-Libraries.md

uxp, 2016-01-05:
I think I've traced this down to the combination of C and C++ sources, which is basically what you said on Aug 21st. 

I'm working inside keyboard/infinity for reference here. This is the full error when I run `make` for that keyboard:
```
In file included from ../../tmk_core/tool/mbed/mbed-sdk/libraries/mbed/api/timer.h:19:0,
                 from matrix.c:4:
../../tmk_core/tool/mbed/mbed-sdk/libraries/mbed/api/platform.h:25:19: fatal error: cstddef: No such file or directory
 #include <cstddef>
                   ^
compilation terminated.
make: *** [build/matrix.o] Error 1
```
But, looking at matrix.c, I see that we want to be including `../../tmk_core/common/timer.h`, _NOT_ `../../tmk_core/tool/mbed/mbed-sdk/libraries/mbed/api/timer.h`. 

Simply removing or renaming `../../tmk_core/tool/mbed/mbed-sdk/libraries/mbed/api/Timer.h` to Timer.hpp, or whatever, I'm able to compile the project and move on.

tmk, 2016-02-17:
I've missed or passed over last post by @uxp, this makes sense totally. Thanks.

Root of the problem is clunky TMK build system but I was quite surprised at OSX filesystem case-insensitiveness, insane! you can curse it :D

I think one of workaround is to fix order of directories in `INCLUDE_PATHS`. Make will find common/timer.h first instead of Timer.h of mbed-sdk with this.

```diff
diff --git a/keyboard/infinity/Makefile b/keyboard/infinity/Makefile
index e740b6a..3e552f2 100644
--- a/keyboard/infinity/Makefile
+++ b/keyboard/infinity/Makefile
@@ -33,9 +33,9 @@ INCLUDE_PATHS = -I.
 #MOUSEKEY_ENABLE = yes
 
 
+include $(TMK_DIR)/tool/mbed/common.mk
 include mbed-infinity.mk
 include $(TMK_DIR)/tool/mbed/mbed.mk
-include $(TMK_DIR)/tool/mbed/common.mk
 include $(TMK_DIR)/tool/mbed/gcc.mk
 
 program: $(OBJDIR)/$(PROJECT).bin
```

closed 2016-02-19 by tmk