tmk/tmk_keyboard #213: HHKB Pro 2 Type-S RN42 Bluetooth mode does not work with Karabiner

ylorn, 2015-05-13:
I want to achieve this effect on my mac:

- When I tap on the right shift (press and release quickly), it will simulate a Command + Space to change the input source from the current one to the previous one.
- When I press the right shift (hold without releasing it), it will still work as a *shift* modifier key.

This is my configuration for Karabiner: https://gist.github.com/ylorn/43b589daf6d621a7fc47

I tried this file with the internal keyboard of my MacBook Pro, it works just as I expected; I tried this file with the HHKB, with your bluetooth board in it, and connect it to my mac using a USB cable, it still works just as I expected; However, when I disconnect the USB cable and turn on the bluetooth switch, it surely does work as a shift modifier key, but not the simulation of Command + Space.

I was wondering if there is anything to do with the Vender ID and Product ID? Because when I connect it using a USB cable, it shows as a keyboard and has a proper Vender ID and Product ID, just like the image below:
[image]
However, when I connect it to my mac using bluetooth, it looks not right:
[image]

And I also tried to achieve this by custom the keymap. 

- First, I replace the right shift with FN3. And I add a fn function:
`[3] = ACTION_MODS_KEY(MOD_LGUI, KC_SPACE)`. It surely do the work of Command + Space, but lose the function of shift as a modifier key.

- Second, I use the following line as a fn function:  `[3] = ACTION_MODS_TAP_KEY(MOD_RSFT, ACTION_MODS_KEY(MOD_LGUI, KC_SPACE))`. This fails on both purposes. 

- Third, I tried this function: 
```c
 [3] = ACTION_MODS_TAP_KEY(MOD_RSFT, KC_FN4),
 [4] = ACTION_MODS_KEY(MOD_LGUI, KC_SPACE),
```
But it can only work as a modifier key.

So these are all I've tried, would you please give me some guide on how to make HHKB work with Karabiner on bluetooth mode? Or if there is any other workaround, please let me know.

tmk, 2015-05-13:
I don't think the VID and PID are problem, I'm not sure how OSX and Karabiner handle BT device though. I can't come up with any cause on that right shift doesn't work on BT at this moment.
THe problem appears only on right shift? did you try righ alt with karabiner instead, for example?

In TMK keymap you can't define functions like second and thrid of your examples.
To realize that setting  on keymap.c you have to write C code in `action_function()` like this. And it is not easy job and not documented well unfortunately.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/hhkb/keymap_hasu.c#L246-L282

Using Karabiner is easier if possible.

ylorn, 2015-05-13:
Just tried with right alt. 
Nope.
Still, works with internal keyboard and HHKB USB mode, but not with the bluetooth mode.

Just out of curiosity, do you use Karabiner as well? Have you ever encounter such problem?

tmk, 2015-05-13:
I don't have OSX :D 
and don't know about Karabinar but seems like you can look into the problem with debugmode.
https://pqrs.org/osx/karabiner/document.html.en#debugmode

tmk, 2015-05-14:
Just skimmed through Karabiner manual,
do you really need inputsource filter? Both "Change Input from English to Chinese" and "Change Input from Chinese to English" have same content, seems to me you can remove 'inputsourcedef' and 'inputsource_only'.

Your configuration looks leagal and it should work with bluetooth keyboard, but I suspsect that Chinese and English 'input method's setting for some reason.

As for Vendor ID and Product ID of the bluetooth keyboard, RN-42 has no method to configure them probably. If this is cause of the problem it is difficult to solve it.

ylorn, 2015-05-15:
Thanks for your reply!

I've always thought that you are using OS X:) Because OS X is natively friendly with Emacs, and I heard that's why every mac coder should use HHKB:)

For the input source filter, I think you're right. I should modify it to get it simpler.

I've already posted a similar [issue](https://github.com/tekezo/Karabiner/issues/403) to the author of the Karabiner. I hope, with his help, we can find out why and get this fixed.

ylorn, 2015-05-16:
We solve the problem. The Karabiner does not allow a device whose Vender ID and Product ID is zero, by default. But we can change it in the settings.
Now everything works as I expected.

Thank you for your help and the wonderful controller board!

closed 2015-05-16 by ylorn