tmk/tmk_keyboard #214: Contributing to the TMK Keymap Editor

daschwa, 2015-05-15:
Hello, 
Is it possible to contribute more keyboards to the [TMK Keymap Editor](http://www.tmk-kbd.com/tmk_keyboard/editor/index.html)? Is the editor under any particular license and is it even a part of this project?

The reason I ask is because I recently adapted the editor to configure the layout and firmware of my own personal keyboard. I would also like to add support for a number of other keyboards besides the HHKB.
Thanks.

tmk, 2015-05-15:
I have to check third party codes I used in that, though,  thinking GPL3. More permissive license is desired?

BTW I've wanted to add ISO layout in the editor for a while but I can't come up with how to define shape of ISO enter key in HTML and CSS :D

daschwa, 2015-05-19:
Yeah, that ISO key would be a little tricky. I wanted to publish my version of the editor under MIT, but if it's under GPL, then I guess I have to go with that.

closed 2015-05-20 by daschwa