tmk/tmk_keyboard #223: LUFA stack using 10-20% CPU in "System Interrupts" (windows)

kobeyu, 2015-05-21:
So I noticed in resmon.exe that "System Interrupts" or "Deferred Procedure Calls and Interrupt Service Routines" was taking 10-20% CPU nearly all the time.

Unplugging my ergodox (with TMK/LUFA stack) made the problem go away.

Out of interest, I recompiled the firmware with the PJRC stack, and the problem went away as well (although the PJRC stack has other issues).

I've made 2 traces I recorded using USBPcap - in both recordings I pressed the windows key a couple times then Ctrl-Ced the recording. You can see the LUFA one is constantly raising interrupts, but the PJRC one only raised a handful.

I don't know how to read USB/HID frames. Any idea what the LUFA firmware is trying to do?

[LUFA recording](https://box.tockli.me/cloud/index.php/s/RghcIVk82kQGPww)
[PJRC recording](https://box.tockli.me/cloud/index.php/s/XqDDyXwbATP1ZSu)

Let me know if you'd rather I raised this on @cub-uanic 's ergodox branch, but he has said in another issue that he hasn't played with the USB stacks.

tmk, 2015-05-21:
Most of  that LUFA data is probably 'debug console data'(Endpoint 4 IN), uneeded empty data seems to be sent every 1ms. `Console_Task` of **protocol/lufa/lufa.c** may be a cause.

Build firmware with commenting out `CONSOLE_ENABLE` line of your Makefile to see the difference of CPU usage.

kobeyu, 2015-05-21:
Commenting out `CONSOLE_ENABLE` seems to produce a firmware which doesn't work at all (well, `Magic-Space-B` got me back to the bootloader, but it didn't work as a keyboard.)

kobeyu, 2015-05-21:
...actually, it works a bit more than that. Layer switching works, and my bootloader keystroke works, and power. But none of the normal keys seem to.

tmk, 2015-05-21:
You are on Windows? If so remove device drivers of your keyboard from Device Manager.
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Build#2-remove-drivers-from-device-managerwindows

kobeyu, 2015-05-21:
Thanks. I'm now running LUFA without the cpu drain, but also without the console. I don't really need the console for day to day use (although I did have an idea to make a layer visualiser using a wrapped hid_listen.exe).

tmk, 2015-05-21:
I'll look into `Console_Task` code later.

BTW, What's your OS actually? I don't see this problem on my Windows 7.

kobeyu, 2015-05-21:
Windows 8.1 Pro. I've got another win 8.1 pro machine at home. I'll try it on that later and see if I get the same behaviour.

tmk, 2015-05-22:
Hopefully fixed at https://github.com/tmk/tmk_keyboard/commit/fdce0c9cc0b7f8e9f1497cae3ea63a6672ceaf71

closed 2015-05-26 by tmk