tmk/tmk_keyboard #225: Adding an ADB mouse's buttons to keycap_<variant>.c

zd9, 2015-05-23:
I have an [ADB Kensington Turbo Mouse](http://ecx.images-amazon.com/images/I/41XXV8PY2ML.jpg). I was able to make left click and cursor movement work with the ADB/USB converter firmware, but I am unsure of how I would add other [Mousekeys](https://github.com/tmk/tmk_keyboard/blob/master/doc/keymap.md#13-mousekey) for the Mouse's buttons to the keymap. Any Ideas?

tmk, 2015-05-23:
You can't change mouse button behaviours in keymap file. Mousekey can be assigned only to keys on keyboard. And currently ADB converter supports only one button.

If you are going to work on developping these supports let me know. I'll be able to explain the detail more.

zd9, 2015-05-23:
I think I can do some development on multiple/programmable mouse buttons. I don't know C too well, but  I can figure it out as I go along.

tmk, 2015-05-23:
Nice.

First, you have to support mouse buttons of ADB mice.

I think you can  find information of ADB mouse protocol among these resources.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/adb.c#L307-L313

Currently the first mouse botton is handled here, probably you can add other buttons code there.
https://github.com/tmk/tmk_keyboard/blob/master/converter/adb_usb/matrix.c#L112-L115

ADB mouse support was added recently by @mek-apelsin, he may know more about these things.
https://github.com/tmk/tmk_keyboard/pull/207

metsarono, 2015-11-16:
I've tried adding support for the secondary mouse button which according to what little I understood of the protocol should use the 8th bit. It seems that all the mouse buttons on the Kensington Turbo are sending the signal for the 16th bit (mouse button one.) 

Would it be possible to hard wire each button to the teensy pin outs directly and set them to mousekey events?

zd9, 2015-11-30:
I think I have the code right, but I have no idea how to implement it on the hardware. I'm using a teensy 2.0.  I know I have to wire each switch on the mouse to the teensy, but I don't know which pins or how to code that part in.

The mouse buttons will be bottom left = btn1, bottom right = btn2, top left = wh_u, top right = wh_d.

**My code (so far):**

in keymap_common.h:

    #define KEYMAP_MOUSE( \
        KF4, \
        KF5, \
        KF9, \
        KFA \
    
    ) { \
        { KC_##KF4 }, \
        { KC_##KF5 }, \
        { KC_##KF9 }, \
        { KC_##KFA }, \
    }

keymap_mouse.c:

    #include "keymap_common.h"
    
    
    const uint8_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
        KEYMAP_MOUSE(
        BTN1,
        BTN2,
        WH_U,
        WH_D
        ),
    };
    
    const uint16_t PROGMEM fn_actions[] = {
    };

tmk, 2015-12-05:
Hardwiring buttons to controller is certainly possible and one of options.

Another option is:
To support mouse buttons more than two we have to implement **Extended Apple Mouse Protocol**.
See https://developer.apple.com/legacy/library/technotes/hw/hw_01.html#Extended and discuss on #247.

closed 2015-12-05 by tmk

zd9, 2015-12-06:
@tmk I get that hardwiring is possible, I just don't know how to do it. How do I add the ports in the code, how do I wire it, etc.

tmk, 2015-12-07:
You will have to check those buttons in matrix_scan() in matrix.c and store state of the buttons in some where part of matrix[] array. As a result, you will be able to remap the buttons in your keymap.c.

You have to know how adb_usb convert works to do this, no document unfortunately and I cannot explain it further here. You have to read my messy codes closely. It'll take time quite a bit.

tmk, 2015-12-07:
If you use you time to implement **Extended Apple Mouse Protocol** instead of hardwire solution I'd appriciate you greatly. Because it allows people to use their multiple-button mouse/trackball easily.

I'd try it if I have Kensington ADB trackball but I cannot get it at reasonable price in Japan anymore.

metsarono, 2015-12-08:
I am very happy to help with testing, but as this is the first time I have ever worked with C my coding support will be quite limited.