tmk/tmk_keyboard #229: How can I redefine the matrix for the Teensy 2.0?

tybenz, 2015-06-12:
I'm trying to build a custom hand-wired board, using the Teensy 2.0 as my board of choice.

I'm doing an 18 x 5 grid.

Here's the `matrix.c` file I'm trying to use: https://github.com/tybenz/switchblade-kb/blob/master/firmware/matrix.c

When I try to compile, I get a complaint about how `PTF0` is not defined. I was trying to use pin references whilst looking at the Teensy 2.0 pinout. What specifically can I do to go from the pinout below to writing a `matrix.c` file? It seems my assumptions about the `matrix.c` pin macros are incorrect.

[image]

tmk, 2015-06-13:
What's PTF0?

tybenz, 2015-06-13:
LOL I don't know. I was seeing `PTB0` in the keymap.c that I was modifying. I took that to mean `PB0` as its labeled on the above pinout. What macro would I use to reference `PF0` on the pinout above?

tmk, 2015-06-13:
You can read datasheet of the MCU to know pin naming scheme and refer to AVR Lib source and document to check definition of those pin name macros.

closed 2015-07-12 by tmk