tmk/tmk_keyboard #232: Umlaute like ä, ö,ü and ß possible?

DonHansDampf, 2015-07-11:
Hey there. I am using a fork of your nice firmware to power my planck. But like the title indicates, i need some umlaute for the geman language. I cant find something about in the Docs, is it somehow possible?

Thanks!

kaansoral, 2015-07-11:
you could temporarily switch the language by sending a key sequence, produce those characters, return back to the original language

with this method, producing any character is possible, but you also have to configure the operating system to have the language set

DonHansDampf, 2015-07-11:
Thanks for the quick answer. I already though about that method, but hoped for a buil-in one. Since i dont really want to be reliant on the OS-language.

kaansoral, 2015-07-11:
It's deemed practically impossible, since keyboards just send keypresses and it's the OS that converts them to characters, as far as I researched, there is no way to tell the OS to output a specific character

I plan on adding triggers for Alt+I, Alt+S etc that will convert them to the special characters of my language - as I don't use the Alt'ed characters and I rarely use my native language while typing

with the infinity controller firmware, this is impossible on 2 levels, one of them is the need for a delay on osx, you can't just instantaneously switch the language and turn back, as it seems

tmk, 2015-07-12:
Unfortunately no OS independent way.

https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#input-special-charactors-other-than-ascii-like-c%C3%A9dille-%C3%87

closed 2015-07-12 by tmk