tmk/tmk_keyboard #238: Can't input the keypad equal.

phdpark, 2015-08-05:
I use adb-to-usb converter on M3501.
But I can't input the kepad equal.
So I have changed the "PEQL" to the "EQL" on "keymap_ansi.c".
And then I can input the equal key on kepad side.
How about others?

tmk, 2015-08-05:
According to HID Usage Table 1.12 that keyoad equal is for AS/400 keyboard, I'm not sure it is supported in your OS. What's your OS?

Probably Mac may support the code while Windows won't.

IIRC current TMK NKRO mode can't send the code make sure using 6RKO mode.

phdpark, 2015-08-05:
I'm using WIN7 Korean.
And my keyboard is Apple Extended Keyboard II.
The original source's firmware works well on my pc except kepad's equal key.
So I modified "keymap_ansi.c"  source.

tmk, 2015-08-05:
Ignore my description about 'AS/400 keypad equal'(0x86) in previous comment, it is not related to this problem.

`PEQL` is 'keypad ='(0x67) and it is supported in Mac acording to HID Usage Table 1.12. And this code can be used in both 6KRO and NKRO mode with TMK firmware.

OK. Now our concern is how Windows handles 'keypad ='. Tried the code on Linux and Windows 7, Linux recognizes as `KP_Equal` and registers `=` on Xorg while Windows doesn't register any character but I could see 'NumpadClear' event occured in AHK window. I don't know why it is recognized as 'NumpadClear' in Windows but you'll be able to use AHK to remap its code.

commit 310a39dd (2016-01-08): adb_usb: Change keycode PEQL to EQL(Fix #238)

Windows doesn't support PEQL(Keypad =)
closed 2015-12-07 by tmk