tmk/tmk_keyboard #241: ACTION_LAYER_TAP_KEY behaving weirdly

p3lim, 2015-08-13:
Using the latest core as of this comment, I've noticed weird behaviors with ACTION_LAYER_TAP_KEY.

I've made a layer (#3, everything is TRNS for testing purposes), and I wanted to do dual roles.
To test this, I wanted to use the LGUI key.

    ACTION_LAYER_TAP_KEY(3, KC_LGUI)

When tapped it fires LCTL and LSFT, not LGUI. Holding it spams these buttons, ignoring the layer completely.

Assigning the key to this:

    ACTION_LAYER_TAP_KEY(3, KC_LALT)

Fires LSFT in the same manner as above (tap is one hit, hold is spamming the key instead of activating layer).

Basically, ACTION_LAYER_TAP_KEY is broken.

p3lim, 2015-08-13:
I used http://www.keyboardtester.com/ for this by the way, it cannot differentiate between left and right modifiers, so for all I know it fires both, didn't test with other keyboard testing applications.

tmk, 2015-08-19:
First you can't use modifier keys with that LAYER_TAP_KEY action. Dual role key is basically comprised of a normal alpha key and a modifier-like key including layer change key. You cannot put together two modifier-like keys on a dual role key.

That being said, yes, that behaviour seem to be weird, I'll have to look into that later. TMK should be fool proof or tamper proof, more tests and fixes are needed clearly.

p3lim, 2015-08-26:
I'd really like for this to be possible though, unless there is some actual reason that it cannot be possible.

tmk, 2015-08-26:
Confirmed.

Problem:
When second argument of `ACTION_LAYER_TAP_KEY` is modifier the action is interpreted as `ACTION_LAYER_MODS`(Momentary switching with Modifiers) wrongly.
https://github.com/tmk/tmk_keyboard/blob/master/doc/keymap.md#35-momentary-switching-with-modifiers

This definition of the action code causes this problem.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_code.h#L74

Workaround:
Do not use modifier with `ACTION_LAYER_TAP_KEY` currently.

With changing definition of the action code this problem can be fixed so that you can give modifier with `ACTION_LAYER_TAP_KEY`. But I don't know this is useful and what you want.

With this fix `ACTION_LAYER_TAP_KEY(3, KC_LGUI)` will change layer when the key is hold and type `LGUI` when they is tapped. Note that 'type' means `LGUI` is registered and unregistered in short time and you won't be able to use it as modifier with other key. Is this what you want?

p3lim, 2015-08-26:
That is exactly what I want.
My use case is that I want to make custom windows commands, for example:

I have my arrow keys on a layer under WASD. In Windows you can move a window around using Winkey + arrow keys. I want to change it so I can do Win+WASD to move the window around. I still want the normal tap to register as LGUI, but have it act as a layer while held.

If you can't (or won't) implement support for modifiers in `ACTION_LAYER_TAP_KEY`, maybe consider adding something like `ACTION_LAYER_TAP_MOD`?

p3lim, 2015-08-26:
My expected layers: [http://www.keyboard-layout-editor.com](http://www.keyboard-layout-editor.com/##@_backcolor=%23444444%3B&@_c=%23b9b8ae&a:7%3B&=Esc&_a:5%3B&=!%0A1&=%2F@%0A2&=%23%0A3&=$%0A4&=%25%0A5&=%5E%0A6&=%2F&%0A7&=*%0A8&=(%0A9&=)%0A0&=%2F_%0A-&=+%0A%2F=&_a:7&w:2%3B&=Backspace%3B&@_w:1.5%3B&=Tab&=Q&=W&=E&=R&=T&=Y&=U&=I&=O&=P&_a:5%3B&=%7B%0A%5B&=%7D%0A%5D&_w:1.5%3B&=%7C%0A%5C%3B&@_a:7&w:1.75%3B&=Caps&=A&=S&=D&=F&=G&=H&=J&=K&=L&_a:5%3B&=%2F:%0A%2F%3B&=%22%0A'&_x:0.25&a:7&w:2&x2:-0.25%3B&=Enter%3B&@_w:2.25%3B&=Shift&=Z&=X&=C&=V&=B&=N&=M&_a:5%3B&=%3C%0A,&=%3E%0A.&=%3F%0A%2F%2F&_a:7&w:2.75%3B&=Shift%3B&@_w:1.25%3B&=Ctrl&_c=%23009bca&w:1.25%3B&=Fn%3Cbr%3EWin&_c=%23b9b8ae&w:1.25%3B&=Alt&_w:6.25%3B&=Space&_w:1.25%3B&=Alt&_c=%23009bca&w:1.25%3B&=Fn%3Cbr%3EWin&_c=%23b9b8ae&w:1.25%3B&=App&_w:1.25%3B&=Ctrl%3B&@_y:0.75%3B&=Win%20+%20L&_g:true%3B&=&=&=&=&=&=&=&=&=&=&=&=&_w:2%3B&=%3B&@_g:false&w:1.5%3B&=Win%20+%20Tab&_g:true%3B&=&_g:false%3B&=Win%20+%20Up&=Win%20+%20E&=Win%20+%20R&_g:true%3B&=&=&=&=&=&=&=&=&_w:1.5%3B&=%3B&@_w:1.75%3B&=&_g:false%3B&=Win%20+%20Left&=Win%20+%20Down&=Win%20+%20Right&_g:true%3B&=&=&=&=&=&=&=&=&_x:0.25&w:2&x2:-0.25%3B&=%3B&@_w:2.25%3B&=&=&=&=&=&=&=&=&=&=&=&_w:2.75%3B&=%3B&@_w:1.25%3B&=&_c=%23009bca&w:1.25%3B&=&_c=%23b9b8ae&w:1.25%3B&=&_g:false&w:6.25%3B&=Win%20+%20Ctrl%20+%20Right&_g:true&w:1.25%3B&=&_c=%23009bca&w:1.25%3B&=&_c=%23b9b8ae&w:1.25%3B&=&_w:1.25%3B&=)

This is with the Win keys working like a doubleshot key, just with a modifier as the key.

tmk, 2015-08-26:
Fair enough. I'll fix this some later.

Use this patch for a while. (This removes 'ACTION_LAYER_MODS' though.)

```
diff --git a/tmk_core/common/action.c b/tmk_core/common/action.c
index 1f15bd0..b7f4f2f 100644
--- a/tmk_core/common/action.c
+++ b/tmk_core/common/action.c
@@ -237,6 +237,7 @@ void process_action(keyrecord_t *record)
         case ACT_LAYER_TAP:
         case ACT_LAYER_TAP_EXT:
             switch (action.layer_tap.code) {
+#if 0
                 case 0xe0 ... 0xef:
                     /* layer On/Off with modifiers(left only) */
                     if (event.pressed) {
@@ -247,6 +248,7 @@ void process_action(keyrecord_t *record)
                         unregister_mods(action.layer_tap.code & 0x0f);
                     }
                     break;
+#endif
                 case OP_TAP_TOGGLE:
                     /* tap toggle */
                     if (event.pressed) {
@@ -519,7 +521,7 @@ bool is_tap_key(keypos_t key)
         case ACT_LAYER_TAP:
         case ACT_LAYER_TAP_EXT:
             switch (action.layer_tap.code) {
-                case 0x00 ... 0xdf:
+                case 0x00 ... 0xe8:
                 case OP_TAP_TOGGLE:
                     return true;
             }
```

p3lim, 2015-08-26:
Much appreciated :+1: 

I suppose I could overwrite the `action.c` file like I did for `action_macro.c` in this commit: https://github.com/p3lim/keyboard_firmware/commit/e5b09eb1980b34ff92323ac4aa5df80ca69a739b#diff-892a36216e118e6964f0337212dd3393R67

p3lim, 2015-08-26:
According to the keyboard tester, shift and control still gets spammed while holding down the dual role key, that is with the patch applied.

This makes using `ACTION_MODS_KEY` with <kbd>Ctrl</kbd> or <kbd>Shift</kbd> impossible on that layer, and any key that would use the <kbd>Win</kbd> key with modifiers does not work neither.
For example, <kbd>Win</kbd>+<kbd>R</kbd> alone is used for showing the Run dialog in Windows, but with <kbd>Ctrl</kbd> and <kbd>Shift</kbd> spammed, it applies those modifiers to it and ends up doing refreshes (basically <kbd>F5</kbd>) instead.

Basically this issue consists of a bug (the spamming when using <kbd>Gui</kbd> as a modifier) and a feature request (using <kbd>Gui</kbd> as a modifier for `ACTION_LAYER_TAP_KEY`).

tmk, 2015-08-26:
Did you try 'make clean'?

I'm confused. Post your real keymap code, I'll try it to see the problem.

p3lim, 2015-08-27:
I do `make clean` before every build.

Everything can be found in [my repo](https://github.com/p3lim/keyboard_firmware), keymap specific stuff in [keymap.c](https://github.com/p3lim/keyboard_firmware/blob/master/keymap.c).

closed 2021-02-01 by tmk

tmk, 2021-02-01:
`ACTION_LAYER_TAP_KEY` works as expected at this time.