tmk/tmk_keyboard #243: ACTION_LAYER_TAP_KEY does not pop mode when released

Miko4, 2015-08-19:
I have my spacebar as a FN-Key, that is mapped to ACTION_LAYER_TAP_KEY(1,KC_SPACE). That works both for wrinting space and for entering layer 1 as expected. I however expect it do return to the previous layer (or just 0) when i release the key, just like for TAP_MOD the modifier key is not pressed anymore when the key is released. I'm not sure If this is a bug or intended. I'd like to make releasing the space key in layer 1 to enter layer 0, but as i understood i only can react to key presses.

I do use the ergodox-fork from https://github.com/benblazak/ergodox-firmware.git so maybe I'm not up-to date.

tmk, 2015-08-26:
Not sure what your problem is. Post your keymap.

Miko4, 2015-08-27:
Here is my keymap.c : https://gist.github.com/Miko4/602f9cec6017d690c2b3

My problem is that once I entered Layer 1 by pressing space I do not leave it by releasing space.

tmk, 2015-08-27:
You have to place TRNS on the place instead of FN4. And you also should place TRNS instead of ENT on layer 1.
Basically you should TRNS on destination layer for layer switch keys and modifier keys.

Miko4, 2015-08-31:
Do I understand it correctly that every layer change key has to stay the same for all higher layers?

tmk, 2015-08-31:
Basically yes.

closed 2015-08-31 by tmk