tmk/tmk_keyboard #248: ACTION_MODS_ONESHOT(MOD_LSFT) doesn't work well / in an fn layer

kaansoral, 2015-09-23:
I have 2 FN layers, activated by keys A and B
When A is activated, pressing B triggers ACTION_MODS_ONESHOT(MOD_LSFT)
When B is activated, pressing A triggers ACTION_MODS_ONESHOT(MOD_LSFT)

So in theory, regardless of which keycap is pressed first, pressing A+B or B+A should trigger ACTION_MODS_ONESHOT(MOD_LSFT)

But this works 50% of the time, also sometimes the LSFT gets stuck, only pressing the LSFT manually gets it unstuck

(I'm using the usb-usb converter, so I'm wondering whether the speed related problem might originate from that, I need to press A+B together for at least 200-300ms for the one-shot to trigger, momentarily tapping them together doesn't work, even though I make sure they are pressed at the same time)

tmk, 2015-09-26:
You have to place the same action(ACTION_MODS_ONESHOT) on both layer when the action is relate to a modifier key(and layer switching key). This is a limitation of current TMK keymap design.

https://github.com/tmk/tmk_keyboard/blob/master/doc/keymap.md#31-momentary-switching
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#modifierlayer-stuck

Keymap like the following doesn't work. Let's think following cases.
1) When A key is pressed and Layer 1 is activated, B is pressed and released, then A is released.
No problem. this works.
2) When A key is pressed and Layer 1 is activated, B is pressed(a), then A is released(b) and B is released(c).
Doesn't work.  At point of (a) LSFT is registered but at (b) Layer 1 is deactivated and it is Layer 0 now, so releasing B key is handled as FN0 action instead of FN1(LSFT). LSFT is stuck!

```
keymaps[][] =  {
                /* A, B */
    [0] = KEYMAP(FN0, FN0),                                                                        
    [1] = KEYMAP(FN1, FN1),                                                                        
}                         
                          
fn_actions[] = {          
    [0] = ACTION_LAYER_MOMENTARY(1),                                                               
    [1] = ACTION_MODS_ONESHOT(MOD_LSFT),                                                           
} 
```

kaansoral, 2015-09-26:
Thanks for the awesome explanation, I understand what the problem is now

But here is my situation:
```
keymaps[][] =  {
                /* A, B */
    [0] = KEYMAP(FN1, FN2),                                                                        
    [1] = KEYMAP(TRNS, FN0), /*A*/
    [2] = KEYMAP(FN0, TRNS), /*B*/                                                                     
}                         

fn_actions[] = {          
    [0] = ACTION_MODS_ONESHOT(MOD_LSFT),                                                            
    [1] = ACTION_LAYER_MOMENTARY(1),
    [2] = ACTION_LAYER_MOMENTARY(2),                                                       
} 
```

So I understood the cause, but I didn't understand the solution, or whether a solution exists

(On the bright side, I can at least make the current setup work by releasing the keys in order)

tmk, 2015-09-26:
No workaround.
New other keymap framework is needed.

tmk, 2017-05-30:
Fixed at 	ba2883f

closed 2017-05-30 by tmk