tmk/tmk_keyboard #250: Eject key missing from HHKB layout

marcaddeo, 2015-09-30:
Just installed the alt controller, which I assume is just flashed with the default hhkb layout but it seems to be missing the Eject key (Fn + f)

tmk, 2015-09-30:
I don't know it is how it works and if it is useful.

Anyway I don't mind adding that key, feel free to send a pull request. Please test it on windows and Mac if possible.

marcaddeo, 2015-10-01:
On OS X you can use Shift + Ctrl + Eject to lock your screen, so I find it useful :)

I will have to look around the repository and learn more before I can submit a pull request. Will take a look though.

marcaddeo, 2015-10-01:
Hmm, after toying around with this for a little while I'm having trouble finding the keycode for eject. I think OS X intercepts it before Karabiner's EventListener can read it... Or something... Not really sure, because even just hitting Fn + f (as well as holding it down for a while) on the original HHKB controller nothing shows up in EventListener.

Do you have a linux box you could run xev on with the original HHKB controller to see what hitting Fn + f sends to the machine? Not sure if this is even a reasonable solution to finding the keycode....

tmk, 2015-10-01:
I got this on xev. I'm not sure what Eject key on HHKB spits out but I'm sure `KC_EJCT`( `KC_MEDIA_EJECT`) is not it.

```
KeyPress event, serial 38, synthetic NO, window 0x8000001,
    root 0x246, subw 0x0, time 1632092085, (-496,231), root:(215,763),
    state 0x10, keycode 198 (keysym 0x1008ffb2, XF86AudioMicMute), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 38, synthetic NO, window 0x8000001,
    root 0x246, subw 0x0, time 1632092197, (-496,231), root:(215,763),
    state 0x10, keycode 198 (keysym 0x1008ffb2, XF86AudioMicMute), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

tmk, 2015-10-01:
OK. I used my USB-USB converter to look into the keycode, it is 0x6F, namely F20. Try it.

marcaddeo, 2015-10-01:
Awesome! Now I just need to learn how to load a new firmware without screwing everything up! :) That shall be tonights project!

tmk, 2015-10-20:
I added document of HHKB original keycodes.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/hhkb/doc/HHKB_keycodes.txt#L22

closed 2015-10-20 by tmk

marcaddeo, 2015-10-20:
@tmk so has this been added to the hhkb layout? Or did you just document it? My battery just came for my BT controller yesterday so I was gonna try and figure this out this weekend.

I didn't see any code changes in the commits, only the documentation. So I assume I still need to create a PR for this.

As for adding the eject key, do I just need to update `keymap_hhkb.c` and change `TRNS` after the `MUTE` in layer 1 to be `EJECT`? Or does the `EJECT` key first need to be defined somewhere as `0x6F`? OR, and more likely, do I just put `F20` there?

persocon, 2015-10-28:
Bro, having the same issue, tried to add F20 and nothing happened :/
also couldn't modify this: https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#apple-keyboard-fn
to work the fn key properly :/

tmk, 2015-10-28:
I'm sure about original HHKB controller spitting out F20 for Fn+F(eject).
Does Fn+F on original HHKB works?

If you understand this correctly and know HID spec you can implements Apple Fn on HHKB hopefully.

persocon, 2015-10-28:
Isn't a matter of just changing the hex? I thought was and didn't worked haha I may be stupid tho.
I just heard about the Fn+F on other places too, I read on some HID spec that F20 is 0x5F and not 0x6F know, and reading your part of the code where the 0x5F is I thought well better not change anything or I will explode this and burn another chip

marcaddeo, 2015-10-28:
@persocon I was able to get it working by setting the `TRNS` after the `MUTE` key to be `EJCT`.

I'm pretty sure that isn't exactly the same as the default HHKB `Fn + f`, but it seems to be working for me.

persocon, 2015-10-28:
@marcaddeo and the default Fn key from mac you were able to make work too? will try that Mute to Ejct

persocon, 2015-10-28:
```EJCT``` worked like a charm :) if you find out how to make the original ```fn``` key work let me know pleeeease!

tmk, 2015-10-28:
Great. `EJCT` keycode can be used on OSX.
BTW, Widnows and Linux seem to ignores it by default.

Updated FAQ entry.
https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#eject-on-mac-osx

marcaddeo, 2015-10-28:
@persocon just to clarify you don't need to change `MUTE` to `EJCT`, you change the `TRNS` (f) that immediately follows the `MUTE` key (d).

persocon, 2015-10-28:
@marcaddeo know the fn key on mac keyboard? turns out it is another HID thing, by default it is the ```APP``` key, but on mac it doesn't work

marcaddeo, 2015-10-28:
@persocon still not 100% sure what you're talking about. Can you give me an example key combination that isn't working, but should be? I haven't had any problems.

persocon, 2015-10-28:
[image]

This key @marcaddeo :) 
and here it is on my keyboard:
[image]

if you take a look on the ```KEYMAP()``` it is the ```APP``` key by default but on mac it does not at all haha

marcaddeo, 2015-10-29:
Oh... I don't know anything about that, this issue was about the Happy Hacking Keyboard Professional 2 so not sure I can help you there :)

persocon, 2015-10-29:
@marcaddeo thanks :) still needs some work on the case tho
the thing about the ```fn``` key is here: https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#apple-keyboard-fn but I have no idea whatsoever on how to implement this :/ haha guess I will add a blank key on the fn place :v

DidierLoiseau, 2016-01-10:
I am also running into this issue as I am trying to mimic the TypeMatrix 2030 which has "eject" on fn+F6.

As mentioned in jackhumbert/qmk_firmware#82, here is the `xev` output from the TypeMatrix eject key:
```
KeyPress event, serial 53, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191628480, (142,442), root:(363,710),
    state 0x10, keycode 169 (keysym 0x1008ff2c, XF86Eject), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 53, synthetic NO, window 0x5200001,
    root 0xd9, subw 0x0, time 191628552, (142,442), root:(363,710),
    state 0x10, keycode 169 (keysym 0x1008ff2c, XF86Eject), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
(which properly works with the default keyboard shortcut in Gnome 3)