tmk/tmk_keyboard #252: PDOT returns delete

boogerlad, 2015-10-07:
rather than the dot. It seems that it's reversed. When not holding shift, pressing PDOT returns delete. When holding shift, PDOT returns the dot. Strangely, it seems to have been working fine before. Here's the code I used https://github.com/boogerlad/tmk_keyboard/compare/ed9766a7d44cd727dbd59008eff4258745ff87cf...master

I'm sure my wiring is correct because I flashed another firmware where instead of PDOT, I used DOT. No shift returned the dot, and shift returns >.

tmk, 2015-10-07:
Both PDOT and DOT works well for me with the latest TMK source. I don't think it is TMK problem.

If your hardware is correct you have to check your codes.

boogerlad, 2015-10-08:
Do you mean check the keycodes returned, or the code that I modified? I linked the diff in the original post and I just changed the mapping. If you mean keycodes, what software would you suggest? The result is the same whether on Windows or Linux. What would you suggest I check on the hardware side?

tmk, 2015-10-08:
I check PDOT and DOT with my own code on my keyboard.
I use xev on Xorg and aquakeytest on Windows to check key input.

On hardware I'd suggest continuity test with DMM?

And skim over FAQ and Issuses on github, you may find somthing useful to debug.

closed 2015-10-20 by tmk

boogerlad, 2017-01-11:
Ahah! I think I have figured out the issue. Numlock!