tmk/tmk_keyboard #253: "xcape" emulation

jzelinskie, 2015-10-10:
This is less an issue more just a discussion of what this feature might look like.

For reference [xcape](https://github.com/alols/xcape) is a project that intercepts your `CTRL` key and makes it act as `ESC` when used alone and `CTRL` when used in combination with other keys. How it works is it actually waits for `CTRL` key to be released before declaring the key-press. Rather than userspace of the computer doing this work, it would be awesome if it could be implemented in firmware of the keyboard. I haven't looked too closely at the structure of this project after a few simple rebindings of keys. How invasive would a patch adding this behavior be?

priyadi, 2015-10-19:
It is already possible: https://github.com/tmk/tmk_keyboard/blob/master/doc/keymap.md#213-modifier-with-tap-keydual-role

jzelinskie, 2015-10-21:
oh awesome, i got it all setup. Thanks!

closed 2015-10-21 by jzelinskie