tmk/tmk_keyboard #254: LED light up and keeps lighted up when keyboard turns on

persocon, 2015-10-17:
Hey, couldn't figure out this one by my self, when I turn the keyboard on (pluging on usb) the two LED turns on and never turn off, then when I hit the combo Capslock+right shift it turns one led off and I can use the capslock, then again the combo nd the other led turns off and switch off the capslock, and again when I hit the combo keys to use capslock one more time it works as expected with the LED, turning on the capslock LED... how do I solve it to just you know start with both LED off and light up when it will be in use?

tmk, 2015-10-20:
Every keyboard can have different hardware and firmware build. We cannot know what your keyboard is at all till you tell us :(

persocon, 2015-10-20:
Sorry, my bad, I thought the LED lib was for all...
well I'm using a phantom keyboard with the keymap_ansi, does that help? :/

tmk, 2015-10-20:
hmm, I don't have the issue you described, so I think it is specfic to phantom firmware or your hardware.

With bpiphany's original phantome fimware, you don't have this issue?

persocon, 2015-10-20:
never tried the original one because of the lack of documentation for a person who lacks skills on C haha
it's so weird this LED bug man :/ 
somehow I think there's something to be around here: on the file matrix.c 
```
void setup_leds(void)
{
    TCCR1A |=      // Timer control register 1A
        (1<<WGM10) | // Fast PWM 8-bit
        (1<<COM1B1)| // Clear OC1B on match, set at TOP
        (1<<COM1C1); // Clear OC1C on match, set at TOP
    TCCR1B |=      // Timer control register 1B
        (1<<WGM12) | // Fast PWM 8-bit
        (1<<CS12);   // Prescaler 256
    OCR1B = LED_BRIGHTNESS;    // Output compare register 1B
    OCR1C = LED_BRIGHTNESS;    // Output compare register 1C
    // LEDs: LED_A -> PORTB6, LED_B -> PORTB7
    DDRB  |= (1<<6) | (1<<7);
    PORTB  &= ~((1<<6) | (1<<7));
}
#endif
```

BUT I have no idea what's going on there haha sorry to bother with that bro :/ really

tmk, 2015-10-21:
I don't have phantom so I tested on other ATMega32u4 board. It works for me on Linux and Windows. May be you are on Mac? perhaps this issue is specific to your OS.
Two LEDs turns on for a sec on startup and then it displays lock state correctly. One of LED is ScrollLock and never get lit but other CapsLock one works well.

In default tmk it has no special bind like "Capslock+right shift", I don't know why that key bind works like that.

BTW, It seems that phantom firmeware seems is not complete and no one maintains it anymore unfortunately. I can't maintain it and I'll have to remove phantom support from my repository some later.
https://github.com/tmk/tmk_keyboard/issues/173

persocon, 2015-10-21:
yeah, I'm on mac, I saw that the Capslock is some sort of fn0 and when you press it you can access like the media keys and the right shift becomes the capslock, know? 
so no fix for that, I will have to live with two leds lit when I plug the keyboard? :p
sad :( if I knew a lot more C I could try maintain and help with those stuff

tmk, 2015-10-21:
ah, right. phantom has keymap like you said, Fn0+right Shift = Capslock.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/phantom/keymap_ansi.h

Yeah, we need someone has phantom hardware, C skill, Mac and free time. it seems to be rarely chance :D

persocon, 2015-10-21:
haha I have all but C skills, if it was like arduino C basic skills :p

tmk, 2015-10-21:
hmm, I read the first message and think about again.
This seems to be rather an issue of tmk core on OSX than phantom specific. But I've not had bug report like this from others yet and I have no Mac to check it unfortunately.

My guess is Linux and Windows send lock key state to keyboard and the keyboard can update its LEDs, but Mac doesn't send the state or phantom misses it for some reason.

Annoying is initial state of phantom LEDs is on, so workaround is LEDs turns off at startup. This is not real fix but probably mitigates the annoying.
```
diff --git a/keyboard/phantom/matrix.c b/keyboard/phantom/matrix.c
index 6c3ae49..afa292c 100644
--- a/keyboard/phantom/matrix.c
+++ b/keyboard/phantom/matrix.c
@@ -55,7 +55,7 @@ void setup_leds(void)
     OCR1B = LED_BRIGHTNESS;    // Output compare register 1B
     OCR1C = LED_BRIGHTNESS;    // Output compare register 1C
     // LEDs: LED_A -> PORTB6, LED_B -> PORTB7
-    DDRB  |= (1<<6) | (1<<7);
+    DDRB  &= ~((1<<6) | (1<<7));
     PORTB  &= ~((1<<6) | (1<<7));
 }
 #endif
```

persocon, 2015-10-21:
Interesting, if you want me to gather some data about stuff like that, let me know what I have to use and how and I will do that :)

zd9, 2015-11-30:
This happens to me on Arch Linux with my phantom, too.

closed 2021-11-11 by tmk