tmk/tmk_keyboard #256: Arduino pro micro/leonardo bootloader jmp

ahtn, 2015-10-18:
To be able to program my arduino pro micro (Caterina bootloader) by keypress with tmk I needed to add a [custom bootloader_jump function](https://github.com/ahtn/tmk_keyboard/commit/773d609bd87d15bb48b3176209c2e2db0b2db178#diff-15c696338fb35b3fa3201e97e19ece2eR112).

Is this something that you would want to add to the tmk_core or is this kind of thing expected to reside in custom projects.

Either way I'll leave this here in case other people are curious how to implement this.

tmk, 2015-10-20:
Thanks. This is useful info for users of the controller.
I'll look into this and add bootloader support for Sparkfun Pro Mircro, Arduino Leonardo and etc.

Bootloader source code for future reference.
Arduino:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/bootloaders/caterina/Caterina.c
Sparkfun:
https://github.com/sparkfun/SF32u4_boards/blob/master/sparkfun/avr/bootloaders/caterina/Caterina.c

tmk, 2021-11-11:
bootloader jump function should work with pro micro (Caterina bootloader).

https://github.com/tmk/tmk_keyboard/blob/37a452f7c96871e1ca166dc29f7692e1334f4b06/tmk_core/common/avr/bootloader.c

closed 2021-11-11 by tmk