tmk/tmk_keyboard #257: Long press function

priyadi, 2015-10-19:
Is it possible to have different behavior when a key is tapped and long pressed? For example, if I tap 1, then it will send 1. But if I press and hold for more than 1 second, then it will send F1.

tmk, 2015-10-20:
Possible. But it is not supported by default, you have to write custom functions yourself in keymap file.

you can discriminate between tap and hold by `tap.count`. When `tap.count > 0` it is a tap, otherwise hold.

This is an example action which uses the `tap_count`. This action is too complex for example but you will see somehting useful.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/hhkb/keymap_hasu.c#L246-L282

priyadi, 2015-10-23:
I have coded this and successfully implemented the feature:

https://github.com/priyadi/tmk_keyboard/blob/master/keyboard/hid_liber/keymap_priyadi.h
https://github.com/priyadi/tmk_keyboard/blob/master/keyboard/infinity/keymap_priyadi.c

However, tapping delay feels too short for this purpose. Is there a way to make tapping delay for this purpose longer (1 second or more), but without modifying the default TAPPING_TERM?

tmk, 2015-10-23:
No. You cannot change the value key by key.

closed 2015-10-26 by tmk