tmk/tmk_keyboard #258: Feature Request: Converter CapsLock NumLock led controls for custom modes

kaansoral, 2015-10-20:
I would like to have:
1) Windows Mode for LGui/LCtrl switching (a) CapsLock
2) Arrow Mode for 60% Arrow switching (b) NumLock

Using my usb-usb controller, I don't use CapsLock or NumLock, yet the leds are there in my 60%, these leds would be utilised manually for these modes

I think both modes would be manually implementable using custom action functions etc. it's not simple, but seems doable, yet there is no way to manually control the leds (at least not that I know of)

I'm requesting this feature

tmk, 2015-10-20:
You should be able to use led_set() of tmk_core/common/led.h to control your Lock indicators. Probably you can implement the features you requested in custom action functions.

These requests are very specific to your preference, they won't support in tmk_core.

kaansoral, 2015-10-20:
(1) and (2) are obviously very specific, but I'm not requesting them, however I'm requesting the led thing, which is actually extremely useful and I'm sure a lot of people would love to have a working solution

And thanks a lot for the swift reply

Out of curiosity, lets say I set the capslock led with led_set, what happens when the actual capslock gets set/unset?

```
void led_set(uint8_t usb_led)
{
    kbd.SetReport(0, 0, 2, 0, 1, &usb_led);
}
```

This seems to be the usb-usb routine, does the keyboard toggle the led itself? (I mean is the state sent, or is there a report sent to toggle?)

tmk, 2015-10-20:
Yeah, LED API should be revised or extended for backlight and etc, sometime, but I have had no good idea yet so far.

`led_set()` sends command to keyboard, not to host, to set/unset indicators on the keyboard, it doesn't change host lock keys state at all.
You can control lock indicators on your keyboard using `led_set()` completely, but note that host also can send command to control the indicators when you press real numlock, capslock keys or etc.

tmk, 2015-12-07:
related to #174

closed 2015-12-07 by tmk