tmk/tmk_keyboard #265: Adding a 'keymap ideas' directory

tofagerl, 2015-11-04:
There are a lot of threads around with individuals keymaps and macros and functions, but I think this should be the proper place for them. 
Ideally there should be both a series of well commented keymap.c-files with both reasoning and tested code, and possible readme-files accompanying each. But hey, just a lot of random stuff we can look through to get some good ideas for how to use the full 32 FN-functions instead of just making boring macros would be better than just the clean standard keymaps in the 'keyboard' dir.

tmk, 2015-11-05:
I want people to contribute their ideas and keymaps to share with others. Wiki would be easier and useful for people to involve this.

tofagerl, 2015-11-05:
Yes, but the wiki is neither forkable or pushable, and requires a bigger 'investment' for each user to share their keymaps.

kaansoral, 2015-11-05:
This is a great idea

I've checked each keymap file I could find a couple of times to spot useful stuff, but the process wasn't easy, there are lots of features of tmk_keyboard, none of them are documented well, read all the documents, it sucks, no offense, the firmware itself rules - an example keymaps directory would act as a documentation on it's own

With tmk's suggestion I implemented an overriden Alt layer for example, while the "Alt" works as "Alt" as expected, Alt+U becomes Ü etc. - I'm sure a lot of people could benefit from such modifications

I currently want to override the NumLock led and use it as a layer indicator, but as I couldn't find any examples that manually activate/deactivate layers, I gave up on the idea for now, tried some layer.c/led.c routines but none achieved what I want

TL;DR: A community curated tmk examples / keymaps directory would be awesome

tmk, 2015-12-07:
Share your keymap idea here!
https://github.com/tmk/tmk_keyboard/wiki/Keymap-examples

closed 2015-12-07 by tmk