tmk/tmk_keyboard #266: Infinity Keyboard doesn't work on OS X; works on Windows and Linux

kevinmost, 2015-11-04:
Note: I am building this firmware on Ubuntu Linux because I can't seem to get the toolchain working properly on OS X.

Tested on a Macbook running 10.10 and a Macbook running 10.11. Neither device will detect the Infinity Keyboard when it's plugged in. Pressing the flash button on the bottom does put the device into bootloader mode, and then `dfu-util` can flash a firmware from OS X successfully.

Flashing the stock [Kiibohd](https://github.com/kiibohd/controller) firmware that came with the Infinity makes the device work on *all* OSes, including OS X. Flashing tmk-keyboard results in a device that works on Linux and Windows, but not OS X.

On OS X, however, it seems like even when the keyboard is unresponsive, it will show up in `lsusb`.

`lsusb` output when the kiibohd firmware is installed:

```
Bus 020 Device 004: ID 05ac:0259 Apple Inc. Apple Internal Keyboard / Trackpad  
Bus 020 Device 002: ID 0a5c:4500 Broadcom Corp. BRCM20702 Hub  
Bus 020 Device 008: ID 05ac:8289 Apple Inc. Bluetooth USB Host Controller  
Bus 020 Device 023: ID 1c11:b04d 1c11 Keyboard - MD1 PartialMap pjrcUSB full  Serial: Clean master - 2015-11-01 17:19:09 -0800
Bus 000 Device 001: ID 1d6b:ILPT Linux Foundation USB 3.0 Bus
```

`lsusb` output when the tmk firmware is installed:

```
Bus 020 Device 004: ID 05ac:0259 Apple Inc. Apple Internal Keyboard / Trackpad
Bus 020 Device 002: ID 0a5c:4500 Broadcom Corp. BRCM20702 Hub
Bus 020 Device 008: ID 05ac:8289 Apple Inc. Bluetooth USB Host Controller
Bus 020 Device 025: ID feed:abed feed mbed  Serial: 0
Bus 000 Device 001: ID 1d6b:ILPT Linux Foundation USB 3.0 Bus
```

Let me know if there's any more info I can provide and I will provide it ASAP. I'd really like to get this working on OS X

kevinmost, 2015-11-04:
So I also found an old binary I had of tmk_keyboard for the Infinity keyboard that I know worked on OS X at one point, and tried flashing it with `dfu-util`, and it still doesn't work. Now I'm getting suspicious that it might be something else?

The result of `dfu-util --list` is: `Found DFU: [1c11:b007] ver=0000, devnum=5, cfg=1, intf=0, alt=0, name="UNKNOWN", serial="mk20dx128vlf5"` When I try to run `dfu-util`:

```
╰─$ sudo dfu-util -S mk20dx128vlf5 -D infinity.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        12652 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
```

tmk, 2015-11-04:
Thanks for the report.
Yeah, someone also reported exact problem on OSX. I guess there is something wrong related to USB3 or Mac OS X. This problem is specific to tmk/mbed protocol, the latest mbed version may fix this, I'll update mbed code some later.

Now flabbergast's tmk/chibios protocol is also available for Infinity60, it probably works on OSX. See this comment.
https://github.com/tmk/tmk_keyboard/issues/261#issuecomment-151000413

kevinmost, 2015-11-04:
I tried the ChibiOS build and while it is detected on OS X, all my keys were transcribed over one to the left! Maybe something wrong with the matrix setup or something?

Also, the ChibiOS was detected on OS X but not Linux! :(

tmk, 2015-11-05:
Ah, my Infinity60 is prototype and has different matrix than production model. My code is configured for it. comment out this line.
https://github.com/tmk/tmk_keyboard/blob/flabbergast-tmk_chibios/keyboard/infinity_chibios/config.h#L55

(Infinity is one of InputClub's brand?, they call their boards as Inifinty 60% and Infinity ergodox, confusing now :D )

tmk, 2015-11-05:
Really sure?
I confirmed on Linux. It should work on Linux.

    Also, the ChibiOS was detected on OS X but not Linux! :(

kevinmost, 2015-11-05:
Commenting that line out makes it work on OS X now, thanks! :+1: 

But yeah, I did test it on Linux and it won't work. I built and flashed the firmware from Linux because, again, I can't get the build setup working right on OS X yet. But after building and flashing from this device, it's not responding as a keyboard... but is responding perfectly on my Mac!

Output of `lsusb` from the Linux device:

```
Bus 003 Device 002: ID 8087:8001 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04ca:7049 Lite-On Technology Corp. 
Bus 001 Device 004: ID 8087:0a2a Intel Corp. 
Bus 001 Device 003: ID 138a:0017 Validity Sensors, Inc. 
Bus 001 Device 002: ID 04f3:012d Elan Microelectronics Corp. 
Bus 001 Device 012: ID feed:6464  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

kevinmost, 2015-11-05:
I can actually build it on OS X when I use the ChibiOS variant! Very exciting. :)

My Linux box still won't let me type with the keyboard, unfortunately. It's Ubuntu 15.04. Maybe it's the 3.x kernel or something?

tmk, 2015-11-05:
I'm also on 15.04 and I confirmed the problem.
To startup keyboard `hid_listen` console is needed for some reason. This problem is related to debug print, I'll look into further.

For a while workaround is disabling console function by commenting out this in Makefile.
```
#CONSOLE_ENABLE = yes   # Console for debug
```

kevinmost, 2015-11-06:
Nice, that worked! Thanks so much for all the help. If anyone else stumbles upon this issue and is looking for a solution, the way to get your Infinity60 working on OS X and Linux (and you can even build on OS X!):

Check out the `flabbergast-tmk_chibios` branch.

`cd tmk_core/tool/chibios`

`git clone -b kinetis https://github.com/flabbergast/ChibiOS.git chibios`

Go back to the `keyboard/infinity_chibios` folder

Edit the `config.h` and comment out `#define INFINITY_PROTOTYPE`

Edit the `Makefile` and comment out `CONSOLE_ENABLE = yes`

Now `make clean && make` and you have a built .bin that can be flashed with `dfu-util`. :)

kaansoral, 2015-11-06:
Thanks a lot for this info, I will use it to convert my infinities to tmk's when I can find some free time

flabbergast is such a life saver too :)

kevinmost, 2015-11-10:
Also should point out, I had some issues with lag in the keyboard once I moved to the ChibiOS build; I thought maybe ChibiOS wasn't as responsive as tmk was or something? But it turns out that if you look in the `config.h` for the `infinity_chibios` folder, there's a line `#define DEBOUNCE 5`. I think that setting that to `#define DEBOUNCE 0` has removed the feeling of "lag" for me. I probably was typing too fast for that setting and it was missing some keystrokes for me.

tmk, 2015-11-13:
OK. I confirm the key drop problem. Slow scan casues it.
See this issue #268 to solve it.

kevinmost, 2015-11-13:
Oh, glad I'm not going crazy! I'm going to remove that line and try it out for a few days and see if it makes it better. I think setting the debounce to 0 did make it better but it's still not perfect, so I'll try what you mentioned in #268

kevinmost, 2015-11-13:
Seems to be working much better, thanks @tmk for pointing me towards that delay line in the ChibiOS firmware. Everything is working fine with it removed and I'm making a lot less typos!

flabbergast, 2015-11-14:
I'll have a look at the original hid_listen/console problem (on a freescale board with a hw debugger) soon.

I also apologise for the needless delay in the main loop - I put it there pretty much during the development just so that the firmware doesn't "needlessly" run a full busy wait loop; I even documented it somewhere, but then forgot about it. When I saw this issue I realised that that's the problem, but you guys are faster ;)

flabbergast, 2015-12-18:
Hopefully fixed the console problem. Now writing to the console queue has 5us timeout - so when it's full, the function returns after 5us.

The root of the problem is (I think) that Linux handles RAW HID endpoints differently than OS X or Windows - the latter two still accept data over them, even if nobody's listening (which meant that the fw could keep pushing and the console output queue never got stuck); but Linux only does this after at least one someone has listened. Hence running hid_listen on linux got the queue "unstuck" and the firmware then could happily continue correctly.

Anyway, another of my silly mistakes.

tmk, 2015-12-19:
Thanks, I'll pull that fix later.
https://github.com/flabbergast/tmk_keyboard/commit/a75834ee98e94cea329aea224a4daaed1a4bf681

Good to know the different behaviours among OSes, it makes sense indeed.
Some users reported that console function on LUFA causes stack in UEFI or failure on resuming from sleep, I'll have to look into that problems.

phoenixeliot, 2016-11-02:
Is this issue still unresolved? I'm having very strange problems with my newly assembled Infinity Ergodox, and they only happen on macOS. Certain keys only update their state if other keys are pressed or released. (eg ; and ' will only update keyup/keydown state if the J, K, or L key is pressed afterward). The keys behave normally on Windows.

closed 2017-05-30 by tmk