tmk/tmk_keyboard #283: Mouse Button 4 and 5 don't work in Bluetooth

jiexi, 2015-12-17:
* Add mouse button 4 and 5 to a keymap
* Compile and flash to HHKB
* In USB mode, test mouse button 4 and 5
* Unplug USB, turn on BT
* In BT mode, test mouse button 4 and 5

tmk, 2015-12-17:
Unfortunately it is limitation of RN-42 module, horizontal wheel is not supported, IIRC.

jiexi, 2015-12-17:
Darn, do you think it would be a good alternative to use other keycode that can be sent over BT, then rebind those keys to mouse clicks in the OS?

tmk, 2015-12-17:
it depends on how you want to use and how your OS recognize the buttons.

In Ubuntu/Xorg(with my configuration) the button4 and 5 are recognize button8 and 9 in Xorg.(confusing but ...) They are used as browser back and forward in Chrome, it is my main usage so alternative setting for me will be to use keycodes like `KC_WWW_BACK` and `KC_WWW_FORWARD`, or macros like `Alt+Left` and `Alt+Right` in my keymap. As for horizontal wheel I would use just `KC_LEFT` and `KC_RIGHT` for alternatives.

If you can use keymap tool on your OS you can use any unused keycodes in you keymap and remap at will on the tools.

closed 2015-12-19 by jiexi