tmk/tmk_keyboard #285: Bounty: Add Model M support

timofonic, 2015-12-20:
Hello.

How much do you want to add Model M support? I might report my models.

i would like to make a project based on your firmware some day, after I get better at studies. I would love to make a franken-modelm (USB+Bluetooth+MSD+USB hub+ARMSBC+NFC+SC+AUDIO+paddle+Power Bank+...).

Kind regards.

timofonic, 2016-01-01:
I also can send you one too.

p3lim, 2016-01-09:
The only boards/controllers that will be natively supported by this project are those that hasu owns himself, any others you can create your own repo/project for.

antonizoon, 2016-07-19:
Looks like someone already made it for you:

http://blog.lmorchard.com/2016/02/21/modelm-controller/

https://github.com/lmorchard/tmk_keyboard/tree/modelm