tmk/tmk_keyboard #287: Tapping: problem on repeating by 'tap and hold'

tmk, 2015-12-26:
Let's say key is configured as `ACTION_MODS_TAP_KEY(MOD_LSFT, KC_BSPC)` and
you type 'ABc' and want to replace 'c' with 'C'.

1. tap the key to remove 'c' with backspace: 'AB' now
2. hold the key to get capital: 'ABC'

If you do 1 and 2 quickly you will get 'Ac' instead of 'ABC'.

https://geekhack.org/index.php?topic=12047.msg1983963#msg1983963

mollat, 2016-01-26:
I have had similar problems with my mod/tap keys. I found myself already making delays, not to run into these mistakes. After reading some code, I think this is not really a bug, this is a feature... ;) But not for me - as my planck keyboard has only 48 keys and - ofcourse - all my 6 modifier keys are configured for single tapping also.

As I am not an experienced C++ programmer, I did not make this feature configurable what should be the normal process. However - it was quite easy to switch it completely off (quick and dirty):

```C++
--- a/tmk_core/common/action_tapping.c
+++ b/tmk_core/common/action_tapping.c
@@ -228,7 +228,8 @@ bool process_tapping(keyrecord_t *keyp)
         if (WITHIN_TAPPING_TERM(event)) {
             if (event.pressed) {
                 if (IS_TAPPING_KEY(event.key)) {
-                    if (!tapping_key.tap.interrupted && tapping_key.tap.count > 0) {
+                    //if (!tapping_key.tap.interrupted && tapping_key.tap.count > 0) {
+                    if (false) {
                         // sequential tap.
                         keyp->tap = tapping_key.tap;
                         if (keyp->tap.count < 15) keyp->tap.count += 1;
```

(just for the record)

Andreas

mollat, 2016-01-27:
I just had another few thoughts about this and - I think the solution could be to let the repeating start with the third tap, so the second tap would not be affected and the 'repeating' feature would still be available. So - maybe I will warm up my very rusty c++ skills in a couple of days...

alexanderjeurissen, 2016-05-22:
Any update on this ? I came across this issue whilst trying to have `Space` and `Left Shift` on the same key using the tap key feature.

this is the function I use.

```
//  FN16 = Shift with tap Space
    [16] =  ACTION_MODS_TAP_KEY(MOD_LSFT, KC_SPC),
```

when I type short words that are separated by a space then often the Left Shift is triggered instead of the tap key.

tmk, 2016-05-22:
No update and no plan in near future.

mollat, 2016-05-23:
I would recommend trying my monkey patch above to combine faster typing speed and action keys on 'enter' or 'space'. 

For me I 'actionized' also both space and both shift keys. To finally keep a (seldom used) repeating space and enter key, I put them additionally onto another layer. See here:

https://github.com/mollat/qmk_firmware/blob/master/keyboard/planck/keymaps/mollat.c

However - there is still a slightly different timing while handling modifyer keys this way in comparison to normal shift or space keys. I did not get behind this yet, but It does not bother me much - and I am used to write quite fast also.