tmk/tmk_keyboard #289: Macro: cannot send mouse key

tmk, 2015-12-28:
I originaly designed that macro can send keycodes only of 0x04-0x73 to save memory space. So I didn't think about sending mouse and media key events in macro much.
https://github.com/tmk/tmk_keyboard/blob/33714e55827d1cf52df1d3ff143f03fa34ad1f06/tmk_core/common/action_macro.h#L38-L55

But `KEY_DOWN` and `KEY_UP` macro should send be able to send mouse, media, system keys(or even Fn?).
https://github.com/tmk/tmk_keyboard/blob/33714e55827d1cf52df1d3ff143f03fa34ad1f06/tmk_core/common/action_macro.c#L40-L59

`register_code()` cannot send mouse key for some reason. It seems be no reason for it cannot send mouse keys.
https://github.com/tmk/tmk_keyboard/blob/33714e55827d1cf52df1d3ff143f03fa34ad1f06/tmk_core/common/action.c#L352-L425

Problem reported:
https://geekhack.org/index.php?topic=41989.msg1985887#msg1985887

Eric-L-T, 2016-01-29:
I think someone has found a solution. See [this issue](https://github.com/jackhumbert/qmk_firmware/issues/96).

Eric-L-T, 2016-02-01:
So did you take a look?

tmk, 2016-02-02:
yes. It is nice workaround for him, but not solution for this Macro issue.
Thanks, anyway. good to know.