tmk/tmk_keyboard #290: Add ability to use strings

JuanPotato, 2016-01-06:
I want to easily be able to send strings like `( ͡° ͜ʖ ͡°)` when I press a key, but it doesnt seem that this is possible without using macros to send the alt codes. Why not add a method to send strings.

tmk, 2016-01-06:
Is that possible? Could you send pull request? I'll add it unless it changes tmk_core too much.

bgould, 2016-01-06:
Seems like this might be difficult to achieve, and probably even more difficult to do it in a way that doesn't depend on the OS of the host computer.

JuanPotato, 2016-01-06:
I know that it is possible as seen [here](http://randomnerdtutorials.com/teensy-username-and-password-auto-filler/), I would attempt to implement it if I had time to look through all the code. It seems simple enough, just send all the characters. Another option would be to add sending unicode characters in macros.

JuanPotato, 2016-01-06:
@bgould Could you explain why?

tmk, 2016-01-06:
It may be helpful to check this.

https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#user-content-input-special-charactors-other-than-ascii-like-cédille-Ç

vovcacik, 2016-01-06:
Probably possible, but not practical at all.

You would need to create macro that sends unicode alt codes, which needs to be enabled on Windows in registry `EnableHexNumpad`. Who knows how this works on other OSes.

Seems like lot of pain with no good usecase.

p3lim, 2016-01-06:
Use macros and a compose software on your OS, I do [this](https://github.com/p3lim/keyboard_firmware/blob/master/keymap.c#L78-L89) for norwegian letters (with a modified macro.c because of #116)

bgould, 2016-01-06:
@JuanPotato The link @tmk posted is what I meant... there is no way to achieve this without making assumptions about the host operating system.

I use this macro to do special characters in X on GNU/Linux:

https://github.com/bgould/costar_tmk_keyboard/blob/master/keymap.c#L81-L106

But this certainly won't work on Windows, and there is no way for me to determine at runtime what OS my keyboard is hooked up to

closed 2016-01-26 by tmk