tmk/tmk_keyboard #293: waiting_buffer_has_anykey_pressed is declared but never used.

obones, 2016-01-07:
Hello,

Thanks for this awesome framework, it makes creating keyboard projects so much easier.
When compiling, I get one warning:

    ../../tmk_core/common/action_tapping.c:327:6: warning: 'waiting_buffer_has_anykey_pressed' defined but not used [-Wunused-function]
     bool waiting_buffer_has_anykey_pressed(void)
          ^

Searching on the web, I found this message: https://geekhack.org/index.php?topic=41989.msg1108557#msg1108557

As it is from 2013, I believe that this was overlooked, but as I understand that this is not a real life issue, I'm creating this issue as a reminder.

p3lim, 2016-01-09:
It's just a warning from the compiler that the function is not used anywhere, the reason the compiler warns about this is so that you could clean up your code.

You can safely ignore this, if the compiler has any serious issues it will just fail.

obones, 2016-01-13:
Yes, I know that it is pretty harmless, but as I'm aiming at zero warning in my various projects, I was reporting this one just in case.

tmk, 2016-01-15:
I can't remember why this function exists there but I don't feel like removing it.

Is there any trick of GCC attribute or somehting to prevent warning except for commenting out?

tmk, 2016-01-26:
I find GCC `unused` attributes but it seems to spoil intention of `-Wunused` at second thought.
https://gcc.gnu.org/onlinedocs/gcc-3.2/gcc/Function-Attributes.html#Function%20Attributes

I'll remove the code then.
https://github.com/tmk/tmk_keyboard/blob/1a02ebcc612e9a9c0d87e02295c7258de3a70ccc/tmk_core/common/action_tapping.c#L327-L333

closed 2016-01-26 by tmk