tmk/tmk_keyboard #296: #error "USART configuration is needed." # error "USART configuration is needed." ^ ../../tmk_core/rules.mk:550: recipe for target 'obj_terminal_bluefruit/keymap.o' failed make

president-obama, 2016-01-15:
I get the follow error when attempting to compile terminal_Bluetooth

fedora terminal_bluefruit] $ make -f Makefile
sh: dfu-programmer: command not found 
sh: dfu-programmer: command not found
-------- begin -------- avr-gcc (Fedora 4.9.3-1.fc23) 4.9.3 Copyright (C) 2015 Free Software Foundation, Inc. This is free software; see the source for copying conditions. There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
mkdir -p obj_terminal_bluefruit Compiling C: keymap.c avr-gcc -c -mmcu=at90usb1286 -gdwarf-2 -DF_CPU=16000000UL -DINTERRUPT_CONTROL_ENDPOINT -DBOOTLOADER_SIZE=512 -DPS2_USE_USART -DPROTOCOL_BLUEFRUIT -DPROTOCOL_PJRC -DPS2_USE_USART -DBOOTMAGIC_ENABLE -DMOUSEKEY_ENABLE -DMOUSE_ENABLE -DEXTRAKEY_ENABLE -DCONSOLE_ENABLE -DCOMMAND_ENABLE -DVERSION=5611f94-dirty -Os -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -fno-inline-small-functions -fpack-struct -fshort-enums -fno-strict-aliasing -Wall -Wstrict-prototypes -Wa,-adhlns=obj_terminal_bluefruit/keymap.lst -I. -I../../tmk_core -I../../tmk_core/protocol -I../../tmk_core/protocol/bluefruit -I../../tmk_core/protocol/pjrc -I../../tmk_core/protocol -I../../tmk_core/common -std=gnu99 -include config.h -MMD -MP -MF .dep/obj_terminal_bluefruit_keymap.o.d keymap.c -o obj_terminal_bluefruit/keymap.o In file included from <command-line>:0:0: ./config.h:65:5: error: #error "USART configuration is needed." # error "USART configuration is needed." ^ ../../tmk_core/rules.mk:550: recipe for target 'obj_terminal_bluefruit/keymap.o' failed make: *** [obj_terminal_bluefruit/keymap.o] Error 1

president-obama, 2016-01-15:
When attempting to compile this i got this error. I was using a teensy 2.0 ++ at90usb1286.. I tested the compilation on the teensy 2.0 and it worked. What changes have to be made to compile this on a at90usb1286 chip.

tmk, 2016-01-15:
I have to  use ``` notation to quote code or error message to retain its format.
<pre>
```
blah blah blah
```
</pre>

Check datasheet of both chips to know diffrence between them and edit `config.h`.

tmk, 2016-01-15:
And edit title to proper one, plz.

president-obama, 2016-01-15:
thanks for the tips i will see If i can get the config.h updated for the at90usb1286.

president-obama, 2016-01-15:
I went over the data sheet both chips are identical in the necessary fields. After review and comparing the config.h file to the usb converter i noticed that the uart ps/2 usart config was missing so no pin was set for clock or data . The missing code from config.h in terminal_bluefruit is below.

...
/* 
 * PS/2 USART configuration for ATMega32U4
 */
#ifdef PS2_USE_USART
/* XCK for clock line */
#define PS2_CLOCK_PORT  PORTD
#define PS2_CLOCK_PIN   PIND
#define PS2_CLOCK_DDR   DDRD
#define PS2_CLOCK_BIT   5
/* RXD for data line */
#define PS2_DATA_PORT   PORTD
#define PS2_DATA_PIN    PIND
#define PS2_DATA_DDR    DDRD
#define PS2_DATA_BIT    
...

Thank you for the direction .

closed 2016-01-15 by president-obama