tmk/tmk_keyboard #297: Layer Switching Problem

Eric-L-T, 2016-01-18:
I've made [a keymap](https://gist.github.com/Eric-L-T/9c3f8093eacca1b905cb) that only requires 1.5 keys for switching layers. However, I've run into a problem.

I have created custom "Shift" layers that reorganize character pairings on keys based on the Classic Dvorak layout. When I go from layer 5 down to layer 0 and then try to type a ~~left bracket~~semicolon after having held the `MO(5)` key the whole time, I instead get a ~~less-than sign~~quotation mark, the character that is paired to the apostrophe by default. I have to lift that key and reengage it as a `MO(1)` key to get my desired character.

Is there a way to work around this? I have been toying around with macros for a while but I haven't made much progress. I am thinking of making the keyboard automatically pretend that the `MO(5)` key was released and pressed again as a `MO(1)` key when it switches layers.

tmk, 2016-01-18:
What's `MO()`, `TO(), F() and M()`? Give us definition of them.

I think you are using diffrent code than tmk and you better ask author of it.

Eric-L-T, 2016-01-18:
Oh, yes, I forgot to tell you that I am using QMK. I asked the owner about my problem, but I think the problem might have to be fixed at a lower level than he can access.

1. `MO()` momentarily switches to the layer whose number is in the parentheses.

1. `TO()` goes to the layer whose number directly follows the opening parenthesis.

1. `F()` references the function whose number is in the parentheses.

1. `MO()` references the macro whose case number is in the parentheses.

tmk, 2016-01-18:
Hmm, what's your problem? And I can't locate `MO(1)` and `MO(5)` in your keymap.
I can't still understand how your keymap works due to its complex. Can you make simpler keymap example to indicate the problem? Pseudo code is enough.

And describe key operations step by step to cuase your problem including all of key press and release.  Like this.
1. press MO(5)
2. press KC_SCLN
3. release MO(5)
4. release KC_SCLN

Eric-L-T, 2016-01-19:
Sorry, I forgot that I had replaced my `MO(1)`, `MO(3)`, `MO(5)` with `F(1)`,  `F(2)`, and  `F(3)`, respectively.

Here is the key sequence with which I am having a problem. Pretend that I am in layer 4. To get to layer 
0, I would press `F(3)` tap `TO(0,1)` and release `F(3)`, correct? However, a problem arises when I try to do this and then immediately type a character on layer 1. For example, when I try to type a semicolon, I would press `F(3)` tap `TO(0,1)`, keep the key that was formerly `F(3)` but should have turned into `F(1)` held, and then tap the key that I think is the semicolon on layer 1. However, the keyboard seems to remain in layer 0, and when I tap that key I instead get a quotation mark.

Eric-L-T, 2016-01-20:
@tmk Would you like additional explanation?

tmk, 2016-01-20:
> Sorry, I forgot that I had replaced MO(1), MO(3), MO(5) with F(1), F(2), and F(3), respectively.
Could you post new keymap file? It would be useful to understand and discuss it.

And could you post like to your real repository? I'd check definition of TO() with its code, didn't get how it works.

tmk, 2016-01-20:
>Here is the key sequence with which I am having a problem. Pretend that I am in layer 4. To get to layer 
0, I would press F(3) tap TO(0,1) and release F(3), correct? However, a problem arises when I try to do this and then immediately type a character on layer 1. For example, when I try to type a semicolon, I would press F(3), tap TO(0,1), keep holding the key that was formerly F(3) but should have turned into F(1), and then tap the key that produces the semicolon on layer 1. However, the keyboard seems to remain in layer 0, and when I tap that key I instead get a quotation mark, the character which is paired by default to the apostrophe in the same position as the semicolon on layer 1.

Sorry, I can't understand or take many time to do due to my English skill, could you rewrite it in format like?:
1. press <key>
2. release <key>

In the end, Do you think this is a bug or design flaw? or just your confusion?

Eric-L-T, 2016-01-20:
The keymap is up-to-date. [Here](https://github.com/jackhumbert/qmk_firmware/blob/master/quantum/keymap_common.h#L176) is what I think is the code for the `TO()` function.

Scenario 1 (No problems)

1. I am in layer 4 and I want to go to layer 0.
1. I press `F(3)`.
1. I tap `TO(0,1)`.
1. I release `F(3)`.
1. I am now in layer 0.

Scenario 2 (Problems)

1. I am in layer 4 and I want to type a character on layer 1.
1. I press `F(3)`.
1. I tap `TO(0,1)` - I am still holding the `F(3)` key on layer 4, which on layer 0 is `F(1)`, which means that I should be in layer 1 momentarily from layer 0.
1. However, when I tap the key which is the semicolon in layer 1, I get a quotation mark (the shifted version of the apostrophe, which is in the same position in layer 0).
1. I have to release `F(3)` and press it again as `F(1)` to get the semicolon.
1. I would like to skip the step above.

I don't think the error is on my part.

tmk, 2016-01-20:
I read it tens times I think I got what you said.

You type keys like and get `?`(Shift+/), right? It is normal and I think your keymap works as you write here.
```
press   F(3) on Layer 4 
press   TO(0,1) on Layer 5
release TO(0,1) on Layer 5
press   S(KC_SLSH) on Layer 0
release S(KC_SLSH) on Layer 0
release F(1) on Layer 0[formly F(3)]
```

If you want to get semicolon ';' on Layer 2, you have to type like this.
```
press   F(3) on Layer 4 
press   TO(0,1) on Layer 5
release TO(0,1) on Layer 5
release F(1) on Layer 0[formly F(3)]
press    F(1) on Layer 0
press   SCLN on Layer 1
release SCLN on Layer 1
release F(1) on Layer 1
```

Note that TMK keymap action occurs only when key state is changed(pressed/released).

Eric-L-T, 2016-01-20:
Can I make it so that pressing `TO(0,1)` changes `F(3)` to `F(1)` without me having to release it and press it again?

tmk, 2016-01-20:
>  I get a quotation mark (the shifted version of the apostrophe, which is in the same position in layer 0).

Weird. You get double quotation? (Shift+'=")

tmk, 2016-01-20:
No, you cant.

Eric-L-T, 2016-01-20:
On a US keyboard that would be `Shift-'`. I think is because `F(3)` is applying Shift to layer 5. Normally this wouldn't be an issue because there are no symbols on layer 5, but it carries over to layer 0 and causes trouble.

Are you sure there is no workaround?

tmk, 2016-01-20:
What 's your keyboar layout on OS? any ISO latin language?

Eric-L-T, 2016-01-20:
I'm using the standard US QWERTY keyboard right now.

tmk, 2016-01-20:
> I think is because F(3) is applying Shift to layer 5. Normally this wouldn't be an issue because there are no symbols on layer 5, but it carries over to layer 0 and causes trouble.

Yes, if you don't release `F(3)`, it'll keep appling the Shift. It is my design.
You still can change its behaviour with editing tmk_core/common/action.c, but it'll affect all other actions.

Eric-L-T, 2016-01-20:
I was able to disable the Shift while still holding F(3) by replacing TO(0,1) with `clear_keyboard`, though. Is there a similar way that I can tell the keyboard to rescan all keys that are pressed using their mappings in layer 0?

tmk, 2016-01-20:
You can write your own code like matrix.c to control matrix scan, maybe?

Eric-L-T, 2016-01-20:
That's a good idea, but my C coding skills are very rudimentary. Unless you are willing to do it, I guess it's time for me to start learning some more.

Eric-L-T, 2016-01-20:
I did a little test and I found out that if a key on the keyboard is pressed when the keyboard is plugged in, the keyboard will start to spam that key. Therefore, fixing my problem might as simple as interrupting and restarting the scanning of the matrix. What do you think?

tmk, 2016-01-20:
I don't think it is related to this issue. you will have to take time to debug your hardware and firmware. and if you find big in my code send PR :)

 I'll close this issue now, but you can reopen if needed.

closed 2016-01-20 by tmk

Eric-L-T, 2016-01-20:
Do you know where I can get help creating the new macro function I need?

Eric-L-T, 2016-01-20:
Hey, I just saw a few functions that I've never seen before: `matrix_power_up()`, `matrix_scan()`, `matrix_power_down()`. What do these do? I think they might help me.