tmk/tmk_keyboard #298: Retrieving Modifier States

Eric-L-T, 2016-01-21:
I want to be able to change the function of a macro by holding down modifier keys. Is there something that I can put into the `if`statement to detect whether they are being pressed?

tmk, 2016-01-21:
You can use this to know current keyboard report including modifiers.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/action_util.h#L27

Eric-L-T, 2016-01-21:
Thank you. Sorry, I am a bit inexperienced in coding. What exactly should I be putting into my `if` statement?

tmk, 2016-01-22:
Like this.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/hhkb/config.h#L40

closed 2016-01-22 by tmk

Eric-L-T, 2016-01-22:
That worked perfectly! Thanks again!

Eric-L-T, 2016-01-25:
@tmk Hello? If this is impossible, please don't hesitate to tell me.

tmk, 2016-01-26:
I think it is possible. Take time to traverse source codes under common directory, you will get idea.

Eric-L-T, 2016-01-27:
.u,u,a.u,.ua 
> .u,.ua.ua.u