tmk/tmk_keyboard #301: Support for new keyboards?

technomancy, 2016-02-02:
I am considering submitting a pull request to add support for the Atreus. However, I recall reading a while back that the plan was to split out support for different keyboards into their own repos.

Are you still accepting pull requests for adding new keyboards?

tmk, 2016-02-02:
Hi,
No, I can't merge new keyboard support currently and I won't probably.
I prefer to have links to project repositories in this wiki page, and I'll organize the wiki page and fix README of tmk_keyboard for people to find those projects easily. Add a link to your repository there.
https://github.com/tmk/tmk_keyboard/wiki/TMK-Based-Projects

technomancy, 2016-02-02:
No problem; thanks for the info.

closed 2016-02-02 by technomancy