tmk/tmk_keyboard #305: Duplicate debug code in ps2_mouse.c

obones, 2016-02-10:
Hello,

At line [88](https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/ps2_mouse.c#L88) in `ps2_mouse.c`, there is a set of lines that print out debug infos related to what's happening on the PS/2 bus.
This code looks very much the same as the one on line [98](https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/ps2_mouse.c#L98) except that the later is surrounded in `IFDEF` and does not output the current time.
This means that when a movement occurs with the mouse, one gets the debug output twice which seems somewhat verbose.
Is this done on purpose? Looking at the indentation, I guess this is a leftover from some debugging session, but blame did not give me much insight here.
I believe the debug code at line 98 should be removed while the one at line 88 should be properly indented and surrounded with an IFDEF.
What do you think?

obones, 2016-02-10:
Here is a pull request: https://github.com/tmk/tmk_keyboard/pull/306

tmk, 2016-02-16:
merged.

closed 2016-02-16 by tmk