tmk/tmk_keyboard #308: LT function does not toggle back from higher layer after button release for most keys

brokenalarms, 2016-02-16:
I'm not sure if this is already covered by #215, but it seems fairly fundamental and easy to reproduce.

For LT(FOO, BAR), it works as expected for BAR keys KC_ENT, KC_UP, even KC_Z (thus not just non-alphanumeric), but for anything in the home row, holding the key just switches to layer FOO and gets stuck there. This means that unless you have a separate toggle switch for that particular layer, you're just stuck there forever....

Can try and fix myself too if you point me in the right direction.

Thanks!

tmk, 2016-02-16:
What's `LT()` ? I can't find it in my repo :(
Maybe you are reporting in wrong repo?

brokenalarms, 2016-02-16:
I sure am. Meant for qmk (ergodox) then accidentally followed your post through from geekhack, they look similar. Sorry! >,

closed 2016-02-16 by brokenalarms