tmk/tmk_keyboard #309: Converter: Add support for XT keyboards with scan code set 1

papodaca, 2016-02-22:
Would be nice to add support for older keyboards with scan code set 1. Soarer's converter exists but is closed source and is difficult to debug.

tmk, 2016-02-22:
Yes, I also want to add XT protocol. Could anyone help?
I don't have XT keyboard to test.

papodaca, 2016-02-22:
I have a board but it is not confirmed working. I really need access to a logic analyzer to see if it is indeed passing clock signals/data correctly. I may be able to write some code for the teensy to do this. Having trouble looking for documentation for how this protocol works, I guess it should be similar to PS/2 but with a different scan code set? if its that simple can't we use the PS/2 code that we have now with a different scan code table? I really don't understand why Soarer never open sourced his firmware.

bgould, 2016-02-23:
It is not exactly the same as PS2.  You can't use exactly the same code.

XT: http://www.kbdbabel.org/signaling/kbd_signaling_pcxt.png
PS2: http://www.kbdbabel.org/signaling/kbd_signaling_ps2.png

Note for instance that XT is not bidirectional, and the clock is different.  I'm not sure how Soarer auto-detects XT vs PS2 but I wonder if it has to do with the different clock rate.  Anyhow here is scan code set 1 for reference:

http://www.computer-engineering.org/ps2keyboard/scancodes1.html

All of my stuff is in storage right now so I have no way to actually help with this, but this should actually be easier to implement than PS/2 and is definitely feasible with a Teensy

papodaca, 2016-02-23:
Ok I have my XT keyboard working with Soarer`s firmware. I think I will be able to help with this, assuming I am able to understand the how PS/2 is implemented in TMK. I still need to read more of the code.

papodaca, 2016-03-02:
Ok I have XT partially working on my 'XT' branch. It currently correctly reads the make codes, have yet to figure out why the break codes are not detected unless the key is depressed for more than 1 second.

commit f37805e6 (2016-06-04): Merge pull request #355 from papodaca/XT

Add converter for XT keyboards, fixes #309
closed 2016-06-04 by tmk