tmk/tmk_keyboard #313: Removal of 'core' branch

tmk, 2016-02-29:
I have plan to remove core branch of tmk_keyboard repo, instead I setup tmk_core repo for the same purpose.  Sorry for this confusion.

I'm going to develop mainly under tmk_core directory(git-subtree'd) in master branch of tmk_keyboard and I'll update tmk_core repo occasionally. So you can think tmk_core repo as more stable, hopefully :D

Eric-L-T, 2016-02-29:
What features are you planning to develop?

tmk, 2016-02-29:
for tmk_core? no specific plan, mainly bug fixes, pull requests and TODOs.
https://github.com/tmk/tmk_keyboard/issues?q=is%3Aissue+is%3Aopen+label%3ATODO

Eric-L-T, 2016-02-29:
It seems to me that you have a lot of work ahead of you! If you need any help, just let me know. I have some great ideas and some basic C skills.

njbair, 2016-03-29:
This change makes sense, but you may want to change the name of this repo from *tmk_keyboard*, to avoid confusion. Something like *tmk_firmware* or even *tmk_collection* might make more sense.

By the way, [GitHub automatically redirects renamed repositories](https://help.github.com/articles/renaming-a-repository/), so no worries about breaking links.

tmk, 2016-03-29:
hm, I don't thnk the name of repo 'tmk_keyboard' is a problem, because this repo embraces my keyboard related firmwares.

But nice to know that how-to. Github help pages are great resources we must check occasionally :D

Eric-L-T, 2016-03-29:
@tmk To which repository do you prefer that I send pull requests? Also, can you please set up a place where people can have you approve of features before they begin working on them?

njbair, 2016-03-29:
@Eric-L-T, he says in the OP here that main development will continue in this repo (tmk_keyboard), and that tmk_core will be considered the "stable" release.

I also like the idea of an approval queue, although I think all that can happen right here in GitHub Issues. Just create an "IDEA" label, which we can use to propose new features. Then if @tmk likes the idea, he just assigns the issue to that person (or whoever wants to take it on) and changes the labels accordingly.

Eric-L-T, 2016-03-31:
@tmk What are your thoughts?

njbair, 2016-04-03:
@eltang I've been thinking about what you said and I agree that it's confusing trying to get involved. See issue #324 for my thoughts on a solution. I'd love your input there.

tmk, 2016-04-04:
@eltang, sorry for late reply.
Let me/us know your repo/branch URL for that change so that I/we can play with. You may want to create new issue here for your change, or we can discuss also at your repo.

eltang, 2016-04-05:
@tmk Sounds good. I think you will like the improvements I've made. By the way, how often can I expect that you will be on GitHub?

tmk, 2016-04-05:
@eltang I usually check notification mails from github in real time manner but I have no time or no idea sometimes, then just forget to reply until rechecking lists. Looking forward to your repo update!

eltang, 2016-04-12:
@tmk I have a quick question. I haven't seen you integrate any of QMK's features. Is there a reason for that?

tmk, 2016-04-13:
@eltang no problem, take time. I sometime check QMK's activities and watch some intresting features, you will see cherry-pick from there in some future.

fredizzimo, 2016-04-13:
@eltang. Here's my view on why the QMK and TMK have got so separated.

1. Github and most open source projects in the world follows a model where contributors send pull requests(or in some cases patches by mail), if they don't have direct access to the main repository. It rarely follows a model where projects randomly goes out to search for forks and pull in changes from that. Besides that model wouldn't work in the long run, because the forks diverges and things become impossible to merge.

   In the case of QMK, I can't see a single pull request from @jackhumbert, but I might have missed some. So he didn't even try to to contribute before starting doing things his own way. I have no idea why he did like that, but it seems like he intended to contribute at some point https://github.com/jackhumbert/qmk_firmware/issues/35, but for a reason or another it hasn't happened.  

1. I think there's a difference difference between philosophies of @tmk and @jackhumbert. Jack has managed to make his QMK firmware a community owned project, where everyone feels happy to contribute. He accepts most pull requests, very soon after they are made.

   Tmk on the other hand seems to want more control of his software, and keep it his own. Therefore many pull requests are not accepted, and some are modified. This probably has a big effect on the willingness to contribute. But on the other hand, the quality will probably be a lot better. 

   For the concrete difference, take a look at this two links https://github.com/jackhumbert/qmk_firmware/pulse/monthly, and https://github.com/tmk/tmk_keyboard/pulse/monthly. For example the QMK firmware has 30 authors, while the TMK firmware has 2. Also in just one month the QMK firmware has accepted almost half the amount of pull requests that TMK has accepted for the whole history.

  All of this is of course not caused by difference in philosophies, the QMK firmware also had the advantage of being the default, "official" firmware for the EZ-Ergodox. Also both models have their own weaknesses and strength and I think some compromise would be needed. 

For me the situation is really sad, @tmk has done a great work of doing all the hard work in the beginning. Work that wouldn't have got down with a community project. And now the QMK firmware is heading in a different direction, where people will forget @tmk totally. 

Therefore I think we should do everything we can to get these two projects combined again. It will probably mean that we should do yet another split, splitting out the hardware specific parts of the core, and the software side, which handles the layout stuff.

This way the development on the "hardware" side could be much more controlled, while the software side more community drive.

Personally I would also like to see, a little bit more control on the layout side, than what seems to be happening on the QMK side. I have no personal experiences with that firmware, but my bet is that things frequently gets broken, because of the number of contributors the frequency of adding new features.

Therefore I would like to see complete unit tests there. To see that all the mapping stuff really works with all the new features that are constantly added. It would also help in developing those features, since you don't need to flash the keyboard in order to test it.

eltang, 2016-04-13:
@fredizzimo That's a very interesting comment. I will have to do some thinking about it. I can say that many of your observations and hypotheses are correct.

njbair, 2016-04-13:
@fredizzimo You've made some really good points here.

I think this project's user base has far outgrown anything that @tmk originally intended. This has created some growing pains as more and more people have been asking for features that meet their specific needs. In the meantime, the Issues/PR tabs are an absolute mess. "Feature request: mouse jiggling," are you serious? From what I can tell, QMK seems more interested in catering to these whims. Meanwhile, TMK has historically focused on quality code and stability, with extensibility as a secondary goal (e.g., macros and functions are there, but not really built out very well yet).

The way I see it, the TMK project's current greatest needs are as follows:

1. Organization/management
2. Better documentation
3. A clear feature roadmap

### Organization/management

TMK needs a clear set of guidelines instructing contributors on where to post issues, which kinds of PRs will be accepted, etc., which can then be referred to by @tmk (and potentially other maintainers) to keep the Issues queue clean. For example, a rule that states that all PRs must reference an existing, open issue which is tagged as ACCEPTED or similar...this will prevent any whining by someone whose PR was rejected after they spent three hours working on a feature nobody wanted.

Meanwhile, pull requests do need to be dealt with more quickly, to avoid the problem of contributed code going stale. This obviously means that testing needs to happen swiftly as well.

This is all a lot of work, but I think the project has enough interested parties to shoulder the load. All it would take is for @tmk to create a GitHub organization for TMK, create a *development* or *community* branch and assign one or more GitHub users as maintainers for that branch. This setup keeps the core *master* branch stable, and still allows hardcore users to access the latest contributed features.

### Better documentation

Some of you may have noticed that TMK's documentation is incomplete. This is a *huge* issue because a feature is worthless if nobody knows about it. But it's not just the scope of documentation which is lacking, but to some degree the language barrier is an issue, since @tmk is ESL. I can speak from experience that the most rapidly-accepted PRs for this project are for the documentation. I recently contributed a rewording and expansion of the Keymap docs, and it was accepted by @tmk the very next day along with a heartfelt thanks.

The problem here, and with every project, is that documentation is not "sexy." People would rather write actual code and implemente really cool new features. But documentation gets forgotten about, or left as a perennial @TODO.

I'd like to humbly suggest that all contributors take a step back from the codebase, and spend some time contributing to the documentation. It's very easy to write in Markdown, and you'll be surprised how quickly you can work through a README section. Even if all you do is clean up a bit of ESL grammar, you're doing a huge service for @tmk and the community, and you're putting down earnest towards the project which lends you credibility and respect the next time you submit a PR.

### A clear feature roadmap

This one may or may not be something others can help with, but it's primarily a task for @tmk. We all need to know what the long-term goals and intentions are for this project. Zooming in from there, contributors need to know which features and changes take priority, ideally in the context of a timeline. Versioning would be a huge help here, as well.

## Next steps

I have outlined some of these ideas in #324, to which @tmk responded favorably. I just think the guy is a bit overwhelmed, and maybe a little burnt out. He's a trailblazer and a "lone ranger" type, who as @fredizzimo stated above, has accomplished something here with the creation of TMK which few other people could have done. Now the project has reached the point where it can either embrace the interest from outside contributors, or remain as-is and eventually see that interest dwindle as those contributors move on to other projects. If @tmk chooses to embrace contributions, here are some suggested next steps:

1. Create a free GitHub organization for the TMK project
2. Transfer ownership of the **tmk_core** repo to the new organization.
3. Create a *community* or *development* branch and invite contributors to serve as maintainers.
4. First assignment for the new maintainers is to manually transfer all pending issues from the existing repo (there should be an API feature to do this, I believe).

I have tremendous respect and admiration for @tmk, which is why I feel compelled to help with this project however I can, as do many others here. I think now the ball is in his court.

eltang, 2016-04-15:
@jackhumbert @IBNobody @NoahAndrews You guys are the main developers of QMK. We would love your input here.

NoahAndrews, 2016-04-15:
I've got several comments, some of which I want to discuss with @jackhumbert before posting here. I'll start by pointing out that most of QMK's external "contributions" are actually just people uploading their personal keymaps for people to use and get ideas from. So the community activity you're looking at, while more active than TMK, is inflated a good deal.

eltang, 2016-04-19:
@tmk @njbair I just saw @njbair's recommendation to change the name of the repo. I was thinking about a name change myself. I think that either `tmk_firmware` or `tmk_keyboard_firmware` will do.

eltang, 2016-04-19:
@tmk I sent you my first pull request of many that will come.

eltang, 2016-04-21:
@jackhumbert Please read this discussion and post your response. I am trying to figure out how I should contribute to both TMK and QMK at the same time.

fredizzimo, 2016-04-23:
Since this is the most recent issue on the subject, I will reply here, but it's really a continuation of https://github.com/tmk/tmk_keyboard/issues/173 and https://github.com/tmk/tmk_keyboard/issues/324. 

I have been thinking about the separation of the core and the keyboard code, and about submodules versus subtrees(With a daytime job as a software architect I could not stop myself from doing it). And here comes my view on it. 

## Background
In https://github.com/tmk/tmk_keyboard/issues/173, @tmk expressed his concerns about not being able to test all the different keyboards, and that he frequently breaks things. As a result he wanted to split the keyboard code (tmk_keyboard, and third-party projects) and the library_code (tmk_core). Initially he tried to have a core branch, which he then decided to replace by subtrees, at least for his own tmk_keyboard project, other projects can use submodules, or a combination. 

## Git based solutions
First let's look at the two Git based solutions, which are the current way of doing things. If you are familiar with them, you can skip the following sections, there's nothing new there. And **the TL;DR version is that both of the solutions have big problems.**

### Submodules
Before I started to work on the support for [Infinity Ergodox](https://github.com/fredizzimo/infinity_ergodox), I didn't really have any experience with submodules. However during the past few months I have really been learning about their use and limitations, as my repository contains 5 submodules on the top level, and one more inside the tmk_serial_link library.  

For me submodules feel quite intuitive, when working alone, if you think of them as separate repositories. You can go into your submodule directory and run normal git commands from inside there. The parent repository stores a reference to the correct version of the subrepository, so you have to remember to make a dummy commit before other people can see the changes. One thing that I didn't realise first is that you can add remotes and fetch from them like for normal repositories, so there's no need to mess with the .gitmodules for local work.

Submodules don't work well when you switch between branches though, you always have to remember to run ´git submodule update´ afterwards. And things get really ugly when you switch between branches that don't have the submodule at all.

If you just want to push your changes to Github things still work reasonably well. You have to fork all repositories to your own Github and push your submodule updates to that. 

When doing pull requests things start to get complicated. You have to 

1. Make pull requests for your submodules first, 
1. Wait for it to be accepted, 
1. Merge the upstream to make sure that you have the same version yourself
1. Commit to your main repository to update the submodule to point to the merged version that the upstream has
1. Then make a pull requests for your main repository

If you want people to pull from your repositories, rather than the upstream ones, you also have to modify the .gitmodules to point to your Github location. But if you also want to contribute to the upstream, you would have to use relative paths.

All this is probably manageable, but very error prone for contributors. There's also a lot of unnecessary work.

For users, I gave the following instructions, which should work for most cases, but still it's probably too complicated for many.

----------

This repository uses submodules, so after you:

* Clone the repository
* Pull a new version
* Switch a branch

You should run

1. `git submodule sync --recursive` (This is not strictly needed, but in case the location of the repositories changes, it's good to do)
1. `git submodule update --init --recursive`

----------

**These are only some of the complications with git submodules, so I can't really recommend them for everyone**, for my repository they have been working OK so far.

### Subtrees
It's probably easier to work with subtrees than submodules, at least for normal users. However the documentation is not good. For example the [Git book](https://git-scm.com/book/en/v2), does not contain any information about subtrees. Subtrees is also not fully officially part of Git. It's a contrib command, which might not be installed on all systems. 

I'm also afraid of merge conflicts as reported [here](http://stackoverflow.com/questions/25294227/git-subtree-pull-merge-conflict) and [here](http://stackoverflow.com/questions/17242063/git-subtree-conflict-when-pulling-from-central-repo). But I haven't tested those cases myself.

**Therefore if we continue to use subtrees, they have to at least be very carefully tested, in order to see that they really work when there are multiple people contributing, and working across many different branches.**

## The Community
### Split of a community that belongs together

The other problem with the chosen solution, besides the technical problems with both submodules and subtrees, is that it fractions the community. Each keyboard type will form it's own community, and all discussions, many of which are really keyboard independent will happen in those project's issue lists. Normal users might not even be aware that the tmk_core library exists.

I'm of course mostly thinking of QMK here, and I can't tell for sure, but I have the feeling that if the project would have been official, and part of tmk_keyboard, then the contributors would have contributed to here, instead  [... cut ...]  @flabbergast is improving in his [branch](https://github.com/flabbergast/whitefox/tree/flabbergast). There are probably many more that I'm not aware of.
Let's improve our processes, make it a team project, and make the TMK firmware completely unbeatable!

Linux, [Google and Facebook](http://paulhammant.com/2014/01/08/googles-vs-facebooks-trunk-based-development/) are other examples of keeping all the code in the same repository.

**If Linus can do it for Linux, we can do it for tmk\_keyboard.**

tmk, 2016-04-23:
I read half way and gave up this engish reading session for today.  I'll read the rest later, it may take weeks probably :) Thanks, good reading.

I checked "TL;DR" section and was surprised at this. I don't know how this bottom line comes yet, I'll know that after reading. looking forward to it.

> So, let's bring all keyboard projects, and the library together into one repository again.

I was working on my TODOs these days according to my plans, I'll keep these going anyway. This is Git repo no source won't be lost and I can recover anything even if I change my mind.

fredizzimo, 2016-04-23:
Don't worry, take your time, my English is not the best either, as it's my second/third language, so that probably makes the text harder to read than it should be.

And of course you don't have to agree with me, you can and should do things the way you think are the best.

flabbergast, 2016-04-25:
I would personally prefer removing tmk_core from the tmk_keyboard repo and use submodules. This removes the confusion as towards which repo should one do pull requests. I think it also simplifies the workflow for developers and people using git. For people not using git, this would just mean downloading and unpacking one more zip file. I don't see a good reason why to try to keep everything in one repo.

I would also argue for using tags in the tmk_core repo. This makes it easy to refer to a particular version of core API in external projects; and one can also generate an URL to download a zip of a particular release from github. Having an established way to refer to particular versions of API {a.k.a. releases} should also make it "easier" to make API changes, since the external projects *should* always refer to a particular release of tmk_core {or if they don't they're expected to keep up with changes in master/develop}.

Finally, I would be for creating one more 'community' repository {e.g. tmk_community} for keyboard projects and keymaps that are not necessarily maintained by @tmk, with more people having write access. tmk_keyboard would contain @tmk's official keyboards and converters.

tmk, 2016-06-22:
Removed `core` branch anyhoo.

closed 2021-02-10 by tmk