tmk/tmk_keyboard #316: Write error for HHKB alt controller

tybenz, 2016-03-02:
Tried this:

```
~/Downloads → dfu-programmer atmega32u4 erase --force hhkb_firmware.hex
~/Downloads → dfu-programmer atmega32u4 flash hhkb_firmware.hex
```

Got:

```
Checking memory from 0x0 to 0x6A7F...  Empty.
0%                            100%  Programming 0x6A80 bytes...
[Device is write protected.
 X  ERROR
Memory write error, use debug for more info.
```

Why can't I write to the HHKB alt controller?

tmk, 2016-03-02:
which doc did you read for this ? let me know so that I can improve it.

what's your OS ? make sure you have right device driver and enough permission to operate the device.

tybenz, 2016-03-02:
I'm on Mac OS X 10.10.5. I used sudo. And it was working. Is there another driver you can point me to?

I followed instructions from here: https://geekhack.org/index.php?topic=71517.0 which led me to here: https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/doc/build.md#download-and-install

tybenz, 2016-03-02:
I'm scared I somehow deleted the bootloader from the controller by using the erase command. Any ideas on how to get this working again?

tmk, 2016-03-02:
No special driver is needed on OSX and Linux. Yes, **sudo** is needed to get permission in most cases.
No worry, you cannot remove bootloader with this process.

closed 2016-03-02 by tmk

tybenz, 2016-03-02:
Ok. But it *is* broken. How can I get it working again?

tybenz, 2016-03-02:
That is, I erased the firmware that was on there and I cannot flash a new firmware, because it keeps telling me:

```
Checking memory from 0x0 to 0x6A7F...  Empty.
0%                            100%  Programming 0x6A80 bytes...
[Device is write protected.
 X  ERROR
Memory write error, use debug for more info.
```

tybenz, 2016-03-02:
Nvm. Got it to work. Tried `make -f Makefile.rn42 dfu` instead of the 3 separate dfu-programmer commands

tmk, 2016-03-02:
ah, no worries :D you didn't broke bootloader. If bootloader is gone away you won't get output like that.

Didn't following commands work? You have to execute them line by line in a row.
```
$ sudo dfu-programmer atmega32u4 erase --force
$ sudo dfu-programmer atmega32u4 flash <your_firmware.hex>
$ sudo dfu-programmer atmega32u4 reset
```

'$' means default shell prompt of bash, 
BTW, your prompt is '→'!? interesting, never seen it :D

    ~/Downloads →

tmk, 2016-03-02:
Great.

tybenz, 2016-03-02:
I tried those 3 commands separately, and it didn't work, but having it run with Make did the trick. Not sure why those are different.