tmk/tmk_keyboard #318: action_t in actionmap

tmk, 2016-03-03:
`action_t` should be used in actionmap instead of `uint16_t`.

This change breaks existent fn_actions definitions in keymap file like: `const uint16_t fn_actions[]`. `uint16_t` have to be replaced with action_t. 

Also keymap should use type name like `keycode_t` instead of 'uint8_t`.

PENDING: due to breaking existent keymap files of users.

```diff
diff --git a/tmk_core/common/action_code.h b/tmk_core/common/action_code.h
index b39c644..c8d3a8c 100644
--- a/tmk_core/common/action_code.h
+++ b/tmk_core/common/action_code.h
@@ -183,7 +183,7 @@ typedef union {
 /* action utility */
 #define ACTION_NO                       0
 #define ACTION_TRANSPARENT              1
-#define ACTION(kind, param)             ((kind)<<12 | (param))
+#define ACTION(kind, param)             { .code = ((kind)<<12 | (param)) }
 
 
 /*
diff --git a/tmk_core/common/actionmap.h b/tmk_core/common/actionmap.h
index cbbcfe9..ac52997 100644
--- a/tmk_core/common/actionmap.h
+++ b/tmk_core/common/actionmap.h
@@ -25,7 +25,7 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 
 
 /* Keymapping with 16bit action codes */
-extern const uint16_t actionmaps[][MATRIX_ROWS][MATRIX_COLS];
+extern const action_t actionmaps[][MATRIX_ROWS][MATRIX_COLS];
 
 
 /* Modified key */
```

njbair, 2016-03-29:
Could you use [typeof](https://gcc.gnu.org/onlinedocs/gcc/Typeof.html#Typeof) to detect this at compile time and cause an #error with instructions for fixing?

I have to think most people who actually update their TMK from Git, would know enough to change a variable type declaration.

eltang, 2016-04-19:
Wouldn't this change make action codes portable across systems? Values could be directly to the members of `action_t` variables.

tmk, 2016-10-20:
Fixed at e5cb8469af8b

closed 2016-10-20 by tmk