tmk/tmk_keyboard #319: Additional Functionality Request: Mouse Jigglin'

redvers, 2016-03-04:
A basic request would be for a minute mouse-move every X minutes.  Purpose is to fool the HostOS into thinking someone is there.  My personal use-case is to work-around nested DPMS bugs which are killing me.

If it's something that's easy then awesome(!).  If it isn't I'll grab a stand-alone one and feel free to close this out.

Thanks for your consideration.

__red__

Eric-L-T, 2016-03-04:
This is very easy to implement if you use [QMK](https://github.com/jackhumbert/qmk_firmware). If you want to stick with TMK, you will have to modify some files in `tmk_core`.

eltang, 2016-04-13:
A quick update: if #323 is merged, you will not have to use QMK.

closed 2016-04-21 by tmk